nordicsemi/Android-Scanner-Compat-Library #1: NullPointerException with ScanFilter setManufacturerData

jcady, 2016-04-08:
Great library!  Encountered one problem.

With the normal ScanFilter object I can make this call to filter only advertisements that match the specific manufacturer ID (while not caring what the actual manufacturer data is).

```
private final ScanFilter mScanFilter = new ScanFilter.Builder()
            .setManufacturerData(MFG_ID, null)
            .build();
```

But making this call with this scanner compat library and running it on 5.0.1 (and possibly others) results in a crash due to the null byte array.

Here's the stack trace:
```
java.lang.NullPointerException: Attempt to get length of null array
            at no.nordicsemi.android.support.v18.scanner.ScanFilter.matchesPartialData(ScanFilter.java:345)
            at no.nordicsemi.android.support.v18.scanner.ScanFilter.matches(ScanFilter.java:302)
            at no.nordicsemi.android.support.v18.scanner.BluetoothLeScannerCompat$ScanCallbackWrapper.matches(BluetoothLeScannerCompat.java:298)
            at no.nordicsemi.android.support.v18.scanner.BluetoothLeScannerCompat$ScanCallbackWrapper.handleScanResult(BluetoothLeScannerCompat.java:248)
            at no.nordicsemi.android.support.v18.scanner.BluetoothLeScannerImplLollipop$ScanCallbackImpl.onScanResult(BluetoothLeScannerImplLollipop.java:131)
            at android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper$1.run(BluetoothLeScanner.java:347)
```

philips77, 2016-04-09:
Hi,
Thanks, I'll investigate it on Monday.
Have a nice weekend.

philips77, 2016-04-11:
Hi, I did a commit: https://github.com/NordicSemiconductor/Android-Scanner-Compat-Library/commit/57f92375a5a1a7a0e8d6b9eb6482d83b4b248820 Could you give it a try before deploying to bintray?

jcady, 2016-04-11:
Tested and working! Thanks for the quick fix!

philips77, 2016-04-11:
Great :)
I'll publish 0.1.2 tomorrow.

philips77, 2016-04-11:
Also.. I wasn't able to do this using the standard API on Nexus 6p with Android 601. It was not crashing, but I was getting all devices. I assume offloaded filtering is chip-relevant.

jcady, 2016-04-11:
Thanks again!  Really useful library, I appreciate the work that went into it.

Hmm, yeah, that must be the case, it seems there is a fair amount that is chip-dependent in the Android BLE stack.  But good to know so that we can be sure to double-check the manufacturer ID when getting the ScanResult.

philips77, 2016-04-12:
Version 0.1.2 has been published on jcenter.
Please, use:
```Groovy
compile 'no.nordicsemi.android.support.v18:scanner:0.1.2'
```

closed 2016-04-12 by philips77

jcady, 2016-04-12:
Thanks Aleksander!