nordicsemi/Android-Scanner-Compat-Library #5: onBatchScanResults continue to receive after stopScan

Jai-GAY, 2016-06-22:
i have logged (can't reproduce, now under heavy test) cases where scanner.stopScan(nordic_scanCallback); has been issued in API 18 (Android 4.3 HTC phone) but the onBatchScanResults (...) still continue receiving (something like that shown below).
 
D/BluetoothAdapter: stopLeScan()
@Override
public void onBatchScanResults(List<ScanResult> results)
D/BLE_com: nordic_scanCallback -> onBatchScanResults sz = 0
D/BLE_com: nordic_scanCallback -> onBatchScanResults sz = 0
D/BLE_com: nordic_scanCallback -> onBatchScanResults sz = 0
D/BLE_com: nordic_scanCallback -> onBatchScanResults sz = 0
...
...
...

Jai-GAY, 2016-08-16:
Finally i can re-produce easily every time i use a Android 5.1.1 lollipop MR1 device.
see log below.

08-16 14:45:52.419 18975-18975/ D/MainActivity: 
08-16 14:45:52.419 18975-18975/ D/MainActivity: 
08-16 14:45:52.419 18975-18975/ D/MainActivity: 
08-16 14:45:52.419 18975-18975/ D/MainActivity: 
08-16 14:45:52.419 18975-18975/ D/MainActivity: 
08-16 14:45:52.458 18975-18975/ D/BLE_com: 
08-16 14:45:52.491 18975-19015/ D/**BluetoothLeScanner**: onClientRegistered() - status=0 clientIf=5
08-16 14:45:52.531 18975-19213/ D/BLE_com: 
08-16 14:45:52.982 18975-18975/ D/BLE_com: nordic_scanCallback -> **onBatchScanResults sz = 1**
08-16 14:45:52.987 18975-18975/ D/BLE_com: **scanner.stopScan**
08-16 14:45:52.988 18975-19213/ D/BLE_com: 
08-16 14:45:52.999 18975-18975/ D/BLE_com: 
08-16 14:45:53.000 18975-18975/ D/BLE_com: 
08-16 14:45:53.003 18975-18975/ D/MainActivity: 
08-16 14:45:53.003 18975-18975/ D/MainActivity: 
08-16 14:45:53.005 18975-18975/ D/MainActivity: 
08-16 14:45:53.005 18975-18975/ D/MainActivity: 
08-16 14:45:53.014 18975-18975/ D/MainActivity: 
08-16 14:45:53.014 18975-18975/ D/MainActivity: 
08-16 14:45:53.014 18975-18975/ D/MainActivity: 
08-16 14:45:53.015 18975-18975/ D/BLE_com: 
08-16 14:45:53.016 18975-18975/ D/BluetoothGatt: connect() - device: F7:73:DE:00:4D:DE, auto: false
08-16 14:45:53.016 18975-18975/ D/BluetoothGatt: registerApp()
08-16 14:45:53.017 18975-18975/ D/BluetoothGatt: registerApp() - UUID=874550ad-3368-4e76-b32e-ac06c744f991
08-16 14:45:53.018 18975-18975/ D/UartService: Trying to create a new connection.
08-16 14:45:53.018 18975-19015/ D/BluetoothGatt: onClientRegistered() - status=0 clientIf=5
08-16 14:45:53.043 18975-18975/ D/MainActivity: 
08-16 14:45:53.044 18975-19232/ D/BLE_com: 
08-16 14:45:53.049 18975-18975/ D/BLE_com: 
08-16 14:45:53.491 18975-18975/ D/BLE_com: **nordic_scanCallback -> onBatchScanResults sz = 0**
08-16 14:45:53.687 18975-19015/ D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=5 device=F7:73:DE:00:4D:DE
08-16 14:45:53.687 18975-19015/ D/UartService: onConnectionStateChange state = 2
08-16 14:45:53.688 18975-19015/ D/UartService: broadcastUpdate 1
08-16 14:45:53.688 18975-19015/ I/UartService: Connected to GATT server.
08-16 14:45:53.688 18975-19015/ D/BluetoothGatt: discoverServices() - device: F7:73:DE:00:4D:DE
08-16 14:45:53.693 18975-19015/ I/UartService: Attempting to start service discovery:true
08-16 14:45:53.696 18975-18975/ D/BLE_com: ACTION_GATT_CONNECTED
08-16 14:45:53.696 18975-18975/ D/BLE_com: 
08-16 14:45:54.002 18975-18975/ D/BLE_com: **nordic_scanCallback -> onBatchScanResults sz = 0**
08-16 14:45:54.512 18975-18975/ D/BLE_com: **nordic_scanCallback -> onBatchScanResults sz = 0**
08-16 14:45:55.018 18975-18975/ D/BLE_com: nordic_scanCallback -> onBatchScanResults sz = 0
08-16 14:45:55.528 18975-18975/ D/BLE_com: nordic_scanCallback -> onBatchScanResults sz = 0
08-16 14:45:56.043 18975-18975/ D/BLE_com: nordic_scanCallback -> onBatchScanResults sz = 0
08-16 14:45:56.545 18975-18975/ D/BLE_com: nordic_scanCallback -> onBatchScanResults sz = 0
08-16 14:45:57.054 18975-18975/ D/BLE_com: nordic_scanCallback -> onBatchScanResults sz = 0
08-16 14:45:57.563 18975-18975/ D/BLE_com: nordic_scanCallback -> onBatchScanResults sz = 0
08-16 14:45:58.073 18975-18975/ D/BLE_com: nordic_scanCallback -> onBatchScanResults sz = 0
08-16 14:45:58.577 18975-18975/ D/BLE_com: nordic_scanCallback -> onBatchScanResults sz = 0
08-16 14:45:59.088 18975-18975/ D/BLE_com: nordic_scanCallback -> onBatchScanResults sz = 0
08-16 14:45:59.308 18975-18996/ D/BluetoothGatt: onSearchComplete() = Device=F7:73:DE:00:4D:DE Status=0
08-16 14:45:59.309 18975-18996/ W/UartService: mBluetoothGatt = android.bluetooth.BluetoothGatt@5013420
08-16 14:45:59.309 18975-18996/ D/UartService: broadcastUpdate 1
08-16 14:45:59.315 18975-18975/ D/BLE_com: ACTION_GATT_SERVICES_DISCOVERED
08-16 14:45:59.315 18975-18975/ D/BLE_com: 
08-16 14:45:59.315 18975-19232/ D/BLE_com: 
08-16 14:45:59.315 18975-19232/ D/BLE_com: 
08-16 14:45:59.331 18975-18975/ D/BLE_com: 
08-16 14:45:59.331 18975-18975/ D/BLE_com: 
08-16 14:45:59.333 18975-18975/ D/BLE_com: 
08-16 14:45:59.333 18975-18975/ D/MainActivity: 
08-16 14:45:59.335 18975-18975/ D/BLE_com: 
08-16 14:45:59.335 18975-18975/ D/BluetoothGatt: setCharacteristicNotification() - uuid: 6e400003-b5a3-f393-e0a9-e50e24dcca9e enable: true
08-16 14:45:59.371 18975-19350/ D/BLE_com: 
08-16 14:45:59.379 18975-18975/ D/BLE_com: 
08-16 14:45:59.595 18975-18975/ D/BLE_com: nordic_scanCallback -> onBatchScanResults sz = 0
08-16 14:46:00.101 18975-18975/ D/BLE_com: nordic_scanCallback -> onBatchScanResults sz = 0
08-16 14:46:00.372 18975-19350/ D/BLE_com: 
[... 48 lines cut ...]
08-16 14:46:14.375 18975-18975/ D/BLE_com: nordic_scanCallback -> onBatchScanResults sz = 0
08-16 14:46:14.882 18975-18975/ D/BLE_com: nordic_scanCallback -> onBatchScanResults sz = 0
08-16 14:46:15.394 18975-18975/ D/BLE_com: nordic_scanCallback -> onBatchScanResults sz = 0
08-16 14:46:15.897 18975-18975/ D/BLE_com: nordic_scanCallback -> onBatchScanResults sz = 0
08-16 14:46:16.413 18975-18975/ D/BLE_com: nordic_scanCallback -> onBatchScanResults sz = 0
08-16 14:46:16.918 18975-18975/ D/BLE_com: nordic_scanCallback -> onBatchScanResults sz = 0
08-16 14:46:17.428 18975-18975/ D/BLE_com: nordic_scanCallback -> onBatchScanResults sz = 0
08-16 14:46:17.936 18975-18975/ D/BLE_com: nordic_scanCallback -> onBatchScanResults sz = 0
[... cut ...]
08-16 14:46:24.051 18975-18975/ D/BLE_com: nordic_scanCallback -> onBatchScanResults sz = 0
08-16 14:46:24.564 18975-18975/ D/BLE_com: nordic_scanCallback -> onBatchScanResults sz = 0
08-16 14:46:25.074 18975-18975/ D/BLE_com: nordic_scanCallback -> onBatchScanResults sz = 0
08-16 14:46:25.594 18975-18975/ D/BLE_com: nordic_scanCallback -> onBatchScanResults sz = 0
08-16 14:46:26.105 18975-18975/ D/BLE_com: nordic_scanCallback -> onBatchScanResults sz = 0
08-16 14:46:26.614 18975-18975/ D/BLE_com: nordic_scanCallback -> onBatchScanResults sz = 0

Jai-GAY, 2016-08-16:
i can't confirm, but i think it could be due to i called BluetoothLeScannerCompat.getScanner().stopScan(nordic_scanCallback); inside the following function.

public void onBatchScanResults(List<ScanResult> results)

closed 2016-08-16 by Jai-GAY

driveorphone, 2017-12-02:
I still have the issue with Android 6, Samsung S5 : calling stopScan(callback) does not stop the scan, and the onBatchScanResults delivery continues forever… Any solution ?

philips77, 2017-12-02:
Did you try stopping it using a handle's post(...) method? Or not from onBatchScanResults() like suggested above?
Are you sure you are using the same callback object to stop?

driveorphone, 2017-12-02:
I have a BluetoothLeScannerCompat scanner object that I'm using to start the scan, then stop it, directly from my scanner class... 
The callback is the class itself, by extending ScanCallback: could that be the issue?

I can show the relevant parts of my code: 

```
public class ScannerBLEDemo extends ScanCallback
{

	private Context context;
	private BluetoothLeScannerCompat scanner;
	public boolean scanning = false;

	public ScannerBLEDemo(Context ctx)
	{
		context = ctx;
		scanner = BluetoothLeScannerCompat.getScanner();
		scanning = false;
	}

	public void startScan(int mode)
	{
		ScanSettings settings = new ScanSettings.Builder()
				.setScanMode(mode).setReportDelay(1000)
				.setUseHardwareBatchingIfSupported(false).build();
		List<ScanFilter> filters = new ArrayList<>();

		ParcelUuid yanService = new ParcelUuid(UUID.fromString(Doppy.YANCAN_SERVICE_UUID));
		filters.add(new ScanFilter.Builder().setServiceUuid(yanService).build());
		
		scanner.startScan(filters, settings, this);
		scanning = true;
	}

	public void stopScan()
	{
		scanner.stopScan(this);
		scanning = false;
	}

	/* SCAN CALLBACK */

	@Override
	public void onBatchScanResults(List<ScanResult> results)
	{
		super.onBatchScanResults(results);
		Timber.i(TAG+" onBatchScanResults called");
	}

	@Override
	public void onScanFailed(int errorCode)
	{
		super.onScanFailed(errorCode);
		Timber.i(TAG+" onScanFailed, errorCode="+errorCode);
	}

	@Override
	public void onScanResult(int callbackType, ScanResult result)
	{
		super.onScanResult(callbackType, result);
		Timber.i(TAG+" onScanResult called");
	}
}
```

philips77, 2017-12-02:
Do you can start and stop from the same thread?
Does it behaves the same way every time, or sometimes it stops correctly?
Did you try setting true in setUseHardwareBatchingIfSupported?

driveorphone, 2017-12-03:
Setting setUseHardwareBatchingIfSupported to **true** worked actually without changing anything else. I was wrongly thinking that using the compat implementation all times wouldn't cause an issue, kind of _"phones that can do the hardest job should also do the easy one"_  but... no.   ;) 

Thank you a lot for your advice @philips77 !