nordicsemi/Android-Scanner-Compat-Library #6: Would be a good idea to have tests covering the library

aldoborrero, 2016-10-20:
It would be a good idea to have tests to cover the classes.

Maybe some tests from the original implementation could be reused and adapted.

For example [ScanFilterTest](https://android.googlesource.com/platform/frameworks/base/+/260a36d/core/tests/bluetoothtests/src/android/bluetooth/le/ScanFilterTest.java)

philips77, 2016-10-24:
Hi @aldoborrero 

Currently I have a lot of things on my plate but I would happily accept any PR. As you wrote in the issue you mentioned here the tests should be pretty much similar to original ones. All help would be appreciated.

Also, the current version number is 0.2.0 but as I haven't got any issues for a long time now I think it should be then changed to 1.0.0.

aldoborrero, 2016-10-24:
Cool @philips77 I'll try to add a PR as soon as I can. Thanks for your input!

philips77, 2016-11-09:
Hi, did you encounter any problems with the lib? Do you thing I should release the version 1.0.0 on jcenter? Do you use it from there, or have cloned it into your project?

aldoborrero, 2016-11-10:
I've been using the lib some time ago and I found nothing wrong with it!

From my side point of view is good and safe to make 1.0.0 release (and if something arises will be minor bugs).

There's already a merge in RxAndroidBle that adds support for this library to use it underneath the reactive code. So if you feel more confident I can battle test even more this.

What I would like to add to next versions is more tests (those from AOSP doesn't cover too much I guess). So we can increase the code coverage of the project.

And... to add a checkstyle guide to standardise code commits if more people wants to collaborate.

philips77, 2016-11-22:
Hi, I've released the version 1.0.0 on jcenter now.

aldoborrero, 2016-11-23:
Thanks @philips77 :)

closed 2017-06-30 by philips77