nordicsemi/Android-Scanner-Compat-Library #9: Which repository

R3n4rD, 2016-11-22:
Hello,

On the Readme.md, you specify to add the following dependencie : 
`compile 'no.nordicsemi.android.support.v18:scanner:1.0.0'`

How ever you don't specify on which repository. I found a version 0.1.1 available on : 
https://mvnrepository.com/artifact/no.nordicsemi.android.support.v18/scanner/0.1.1
But no 1.0.0. Could you update the readme.md ?

Thanks.

Regards,
Clement.

philips77, 2016-11-22:
Yeah.. this is because the readme update went before the actual release. I'll do the release ASAP. It will be 1.0.0 which is the same as before + unit tests.

philips77, 2016-11-22:
Hi, I've published the 1.0.0 release now on bintray: https://bintray.com/nordic/android/no.nordicsemi.android.support.v18%3Ascanner. It should be available on jcentera and maven now.
Sorry for the delay!

closed 2016-11-22 by philips77

R3n4rD, 2016-11-22:
Thanks a lot :) 
I will have a try. The thing is, i'm facing a 133 GATT issue with a SamsungA5 which seems "casual" on the internet but i'm still trying to figure out why. (gatt operations in uiThread, disconnecting and closing gatt, calling refresh method on disconnection...).

philips77, 2016-11-22:
You could try to modify the connection parameters by sending a request from the peripheral after the phone connects to it. We had a case when it did help (our nrf5-sdk-for-Eddystone project on Github).

philips77, 2016-11-22:
I will try to find out more details about it as it is not me who wrote this sdk.