nordicsemi/Android-Scanner-Compat-Library #10: How to include this library in project where min SDK < 18?

pozix604, 2016-12-19:
I'd like to include Android-Scanner-Compat-Library in a project where min SDK < 18 so that if the device SDK is < 18, then I wouldn't use BLE capabilities, and otherwise I would use it per this library.

However, the manifest requires min SDK 18. Is there any way around this?

Jai-GAY, 2016-12-28:
are you saying you will use use classic profile for operation if below < 18 (which is totally different ball game) ?

Just wondering why a developer would like to do that since hardware is tightly coupled with android API 18 for BLE to be ready. furthermore, the market share below 4.3 is so small to even consider additional effort.

[Android 4.3 (API level 18) introduces built-in platform support for Bluetooth Low Energy (BLE) in the central role and provides APIs that apps can use to discover devices, query for services, and transmit information.](https://developer.android.com/guide/topics/connectivity/bluetooth-le.html)

[This section provides data about the relative number of devices running a given version of the Android platform.](https://developer.android.com/about/dashboards/index.html)

philips77, 2016-12-28:
As far as I understand on devices below API 18 the Bluetooth related features could be disabled.

pozix604, 2016-12-29:
How do I add this library to a project where min SDK  is less than 18?

How I would use this library is as follows:
- If at runtime, I detect that SDK < 18, then I would not use any BLE features
- If at runtime, I detect that SDK >= 18, then I would use BLE, including this library

Right now, if I simply add this library to a project, it requires my project to have min SDK >= 18.

pozix604, 2016-12-29:
Figured it out. Use the following in the manifest file:

`    <uses-sdk tools:overrideLibrary="no.nordicsemi.android.support.v18.scanner"/>
`

philips77, 2016-12-29:
Great! Good to know. I'm closing the issue then.

closed 2016-12-29 by philips77