nordicsemi/Android-Scanner-Compat-Library #11: SCAN_MODE_OPPORTUNISTIC crashes on Lollipop

nielsz, 2017-01-10:
Samsung S5, on Android 5.0.
```
new no.nordicsemi.android.support.v18.scanner.ScanSettings.Builder()
  .setScanMode(ScanSettings.SCAN_MODE_OPPORTUNISTIC)
  .setUseHardwareBatchingIfSupported(false)
  .build();

```

```java.lang.IllegalArgumentException: invalid scan mode -1
at android.bluetooth.le.ScanSettings$Builder.setScanMode(ScanSettings.java:213)
at no.nordicsemi.android.support.v18.scanner.BluetoothLeScannerImplLollipop.toImpl(BluetoothLeScannerImplLollipop.java:186)
at no.nordicsemi.android.support.v18.scanner.BluetoothLeScannerImplLollipop.startScanInternal(BluetoothLeScannerImplLollipop.java:70)
at no.nordicsemi.android.support.v18.scanner.BluetoothLeScannerCompat.startScan(BluetoothLeScannerCompat.java:112)
```
It should fallback to a mode that it does support, which would probably be `SCAN_MODE_LOW_POWER`.

philips77, 2017-01-10:
True. Could you, please, make a PR? I don't have time right now to handle it...

nielsz, 2017-01-13:
My colleague doen't agree what the fallback behaviour should be, as the power consumption on older devices will be a lot higher. Is `SCAN_MODE_LOW_POWER` the best alternative? We could also make a fallback param.
`setScanMode(ScanSettings.SCAN_MODE_OPPORTUNISTIC, ScanSettings.SCAN_MODE_LOW_POWER)`

Anyway, this scenario is very low level for us, and we probably won't need it for some months. 
As I'm swarmed myself I can't take this up.

philips77, 2017-01-13:
That's also true. Have a look at the PDF here: https://devzone.nordicsemi.com/blogs/1046/what-to-keep-in-mind-when-developing-your-ble-andr/
Scanning with low power scans for 500 ms every 5 sec, so from the battery point of view it's quite a lot, but when you really need to find the device it's super small window and you need to make sure the advertising interval is OK.