nordicsemi/Android-Scanner-Compat-Library #12: Caught a RuntimeException from the binder stub implementation

Jai-GAY, 2017-02-20:
getting an unusual RuntimeException if i code this way using version no.nordicsemi.android.support.v18:scanner:1.0.0. Can I use it this way ?
scanner.startScan(**_null_**, settings, nordic_scanCallback);

`02-20 13:08:53.919 12379-12513/ D/BluetoothLeScanner: onClientRegistered() - status=0 clientIf=5
02-20 13:08:53.923 12379-12513/ W/Binder: Caught a RuntimeException from the binder stub implementation.
                                                                     java.lang.SecurityException: Need ACCESS_COARSE_LOCATION or ACCESS_FINE_LOCATION permission to get scan results
                                                                         at android.os.Parcel.readException(Parcel.java:1627)
                                                                         at android.os.Parcel.readException(Parcel.java:1579)
                                                                         at android.bluetooth.IBluetoothGatt$Stub$Proxy.startScan(IBluetoothGatt.java:772)
                                                                         at android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper.onClientRegistered(BluetoothLeScanner.java:324)
                                                                         at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:56)
                                                                         at android.os.Binder.execTransact(Binder.java:453)

02-20 13:09:01.213 12379-12574/ V/RenderScript: 0x7f5fe7e000 Launching thread(s), CPUs 8
02-20 13:09:01.231 12379-12379/ I/Timeline: Timeline: Activity_idle id: android.os.BinderProxy@abbc528 time:42374438`

philips77, 2017-02-20:
Yes, you may, but as the log suggests, you need to request (and be granted, as this is a runtime permission) one of the location permissions to be able to scan on Android 6+.

Jai-GAY, 2017-02-20:
@philips77 could you share besides the 2 following permissions, which permission are you referring to ?

<uses-permission android:name="android.permission.BLUETOOTH"/>
<uses-permission android:name="android.permission.BLUETOOTH_ADMIN"/>

Jai-GAY, 2017-02-20:
noted.

pushpender311, 2017-06-28:
Go to App Info -> Permissions -> and then allow Location permission.

Problem Solved..!!

Jai-GAY, 2017-06-29:
@pushpender311, great, it is solved, default is OFF state, now i have to think how to make it default ON. thanks.

closed 2017-06-30 by Jai-GAY

Palatis, 2017-07-17:
you should ask for permission by using Activity.requestPermissions() or Fragment.requestPermissions(), and start whatever that needs permission in onRequestPermissionResult().