nordicsemi/Android-Scanner-Compat-Library #13: Cannot find devices on SAMSUNG some times

chengjiamei, 2017-07-03:
Cannot find devices on SAMSUNG some times

Jai-GAY, 2017-07-03:
there is no code on how you did the scanning, could you have accidentally turn on the scan filter ?
could it be that other devices already connected and not yet disconnect before SAMSUNG start scanning ?

scanner.startScan(filters, settings, nordic_scanCallback);

hope it is helpful.

chengjiamei, 2017-07-03:
Thank you for response,and i confirm that the device  had not connected,follow is my code
`final BluetoothLeScannerCompat scanner = BluetoothLeScannerCompat.getScanner();
        final ScanSettings settings = new ScanSettings.Builder()
                .setScanMode(ScanSettings.SCAN_MODE_LOW_LATENCY).setReportDelay(1000).setUseHardwareBatchingIfSupported(false).build();
        List<ScanFilter> scanFilters = new ArrayList<>();
        if(mIsScanning) {
            cancelLeScan();
        }
        if(scanCallback == null) {
            scanCallback = new MyScanCallBack();
        }
        scanner.startScan(scanFilters, settings, scanCallback);
        mIsScanning = true;
        mHandler.postDelayed(new Runnable() {
            @Override
            public void run() {
                if (mIsScanning) {
                    cancelLeScan();
                }
            }
        }, SCAN_DURATION);`
There are any problem?Thank you again.

philips77, 2017-07-03:
Try passing null as filters, instead of an empty array.
Are you sure you have and granted the location permission (coarse it fine) if using Marshmallow or newer?

chengjiamei, 2017-07-03:
I am sure i had granted the location permission and i tested on Android 6.0. I will change the param  according to your advice and test again.Thank you very much!

chengjiamei, 2017-09-18:
```
``09-18 09:46:41.677 19958-19958/com.isport.isporttracker D/BluetoothLeScanner: Start Scan
09-18 09:46:41.678 19958-19958/com.isport.isporttracker D/BluetoothAdapter: STATE_ON
09-18 09:46:41.679 19958-19958/com.isport.isporttracker D/BluetoothAdapter: STATE_ON
09-18 09:46:41.680 19958-19958/com.isport.isporttracker D/BluetoothAdapter: STATE_ON
09-18 09:46:41.681 19958-19958/com.isport.isporttracker D/BluetoothAdapter: STATE_ON
09-18 09:46:41.787 19958-19971/com.isport.isporttracker D/BluetoothLeScanner: onClientRegistered() - status=0 clientIf=5 mClientIf=0
09-18 09:46:42.674 19958-19958/com.isport.isporttracker E/BleService: onBatchScanResults  size = 0
09-18 09:46:43.677 19958-19958/com.isport.isporttracker E/BleService: onBatchScanResults  size = 0
09-18 09:46:44.679 19958-19958/com.isport.isporttracker E/BleService: onBatchScanResults  size = 0
09-18 09:46:45.681 19958-19958/com.isport.isporttracker E/BleService: onBatchScanResults  size = 0
09-18 09:46:46.684 19958-19958/com.isport.isporttracker E/BleService: onBatchScanResults  size = 0
09-18 09:46:47.687 19958-19958/com.isport.isporttracker E/BleService: onBatchScanResults  size = 0
09-18 09:46:48.690 19958-19958/com.isport.isporttracker E/BleService: onBatchScanResults  size = 0
09-18 09:46:49.693 19958-19958/com.isport.isporttracker E/BleService: onBatchScanResults  size = 0
09-18 09:46:50.696 19958-19958/com.isport.isporttracker E/BleService: onBatchScanResults  size = 0
09-18 09:46:51.698 19958-19958/com.isport.isporttracker E/BleService: onBatchScanResults  size = 0
09-18 09:46:51.799 19958-19958/com.isport.isporttracker D/BluetoothAdapter: STATE_ON
09-18 09:46:51.801 19958-19958/com.isport.isporttracker D/BluetoothAdapter: STATE_ON
09-18 09:46:51.803 19958-19958/com.isport.isporttracker D/BluetoothAdapter: STATE_ON
09-18 09:46:51.805 19958-19958/com.isport.isporttracker D/BluetoothAdapter: STATE_ON
09-18 09:46:51.805 19958-19958/com.isport.isporttracker D/BluetoothLeScanner: Stop Scan
```

the size that return in onBatchScanResults always   is zero, even on-off bluetoothadapter.
the scan method is 
```final BluetoothLeScannerCompat scanner = BluetoothLeScannerCompat.getScanner();
        final ScanSettings settings = new ScanSettings.Builder()
                .setScanMode(ScanSettings.SCAN_MODE_LOW_LATENCY).setReportDelay(1000).setUseHardwareBatchingIfSupported(false).build();
        scanner.startScan(null, settings, scanCallback);
        mIsScanning = true;
        mHandler.postDelayed(new Runnable() {
            @Override
            public void run() {
                if (mIsScanning) {
                    cancelLeScan();
                }
            }
        }, SCAN_DURATION);
```

I tested on SAMSUM S7 and S6

philips77, 2017-09-18:
Did you try to enable Location in Android drop down menu? Some devices require it to be ON. Doesn't matter what accuracy you set. Also, you may try using https://github.com/NordicSemiconductor/Android-Scanner-Compat-Library/blob/master/scanner/src/main/java/no/nordicsemi/android/support/v18/scanner/ScanSettings.java#L401 with false, as batching fails on some devices. This will, however, impact battery usage.

chengjiamei, 2017-09-18:
Thank you first, i am sure that i had enable Location and make it be ON. And i used setUseHardwareBatchingIfSupported  with false, but it does not work too .

farruxx, 2017-10-27:
Same issue with stock android 7.1 frimwares on xperia

trongkhanh, 2018-10-10:
BluetoothLeScanner on android 6.0.1 (ss s6 edge) cannot found device android >= 6.0.1

i had enable permission for App as following:
uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
uses-permission android:name="android.permission.BLUETOOTH" />
uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
uses-permission android:name="android.permission.SYSTEM_ALERT_WINDOW" />

philips77, 2019-01-11:
Hello,
Did you manage to solve the issue?
Did you requested Location runtime permission? You may grant it Settings -> Applications -> Your app -> Permissions -> switch the Location switch to ON and try scanning.

philips77, 2019-01-31:
HI, did you manage to solve the issue?

closed 2019-02-15 by philips77

Strife91, 2019-03-25:
> Did you try to enable Location in Android drop down menu? Some devices require it to be ON. Doesn't matter what accuracy you set. Also, you may try using [...]

Might I ask why some devices requires it to be on and others not?

philips77, 2019-03-25:
Well... it should be required in all phones. The Location setting is a master switch for all location based features of the phone.
But some manufacturers decided to do otherwise. As Android is open source, they are free to do so. Actually, Samsung now also requires it in Android Pie+.

Strife91, 2019-03-27:
> Well... it should be required in all phones. The Location setting is a master switch for all location based features of the phone.
> But some manufacturers decided to do otherwise. As Android is open source, they are free to do so. Actually, Samsung now also requires it in Android Pie+.

I just thought it was strange because only the OnePlus 3 of the 6 different devices I've tried requires it to be on, even the Nexus 5X does not even though it is vanilla Android, so it's strange that google doesn't enforce it themselves.

And I'm not required to have location enabled when I use bluetooth headsets even though they are bluetooth 4.1.

I just want some clarity here. 

And what do you think is the best solution? I don't want to ask the user to enable location on devices that does not require it to be on. So is there some kind of check I can do to see if location must be enabled on this particular device?

philips77, 2019-03-27:
Location is required only for scanning Bluetooth LE devices. Your headset is Classic Bluetooth and perhaps already paired.

Check what we do in nRF Blinky. We start scanning initially showing a warning to enable Location, but if we find a device despite it being disabled, we save internally that the phone doesn't need it and no longer allow it.