nordicsemi/Android-Scanner-Compat-Library #14: Scanning Problem in API 6.0.1

DLeague9, 2017-07-06:
hello. i'm using your library in my project. Its working perfectly in API 5, But when i test it on 6.0.1 **ScanCallBack** didn't called. 
i'm using **scanner.startScan(mScanCallback)** method and **onScanResult** never triggered. 
Hope you will reply soon and guide me through. thanks

philips77, 2017-07-06:
On Marshmallow+ you have to request Location permission (coarse or fine). It enable it in settings->apps->your app->permissions->location. Please read about new permission scheme on Android 6.

Jai-GAY, 2017-07-06:
May be you have to take a look at the scanner handler you have acquired ?

instead of getting it from the BluetoothAdapter, acquire the scanner instance using:
BluetoothLeScannerCompat scanner = **BluetoothLeScannerCompat.getScanner()**;
You also need to change the packets for ScanSettings, ScanFilter and ScanCallback classes to:
no.nordicsemi.android.support.v18.scanner

hope it helps.

DLeague9, 2017-07-06:
i added **<uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />** in manifest. Nothing worked.
i'm not using BluetoothAdapter. Is that an issue?
My scan method below:
```
private void EstablishConnection(){
        BluetoothLeScannerCompat scanner = BluetoothLeScannerCompat.getScanner();
        scanner.startScan(mScanCallback);
    }
```
Callback methods:
```
private final no.nordicsemi.android.support.v18.scanner.ScanCallback mScanCallback = new no.nordicsemi.android.support.v18.scanner.ScanCallback() {
        @Override
        public void onScanResult(int callbackType, no.nordicsemi.android.support.v18.scanner.ScanResult result) {
            if (result!=null) {
                BluetoothDevice device = result.getDevice();
                if("BLE_PRO".equals(device.getName())){
                    Log.i("DeviceFound: ", device.getAddress());
                    BluetoothLeScannerCompat scanner = BluetoothLeScannerCompat.getScanner();
                    scanner.stopScan(mScanCallback);
                    connect(device);
                }
            }
        }

        @Override
        public void onBatchScanResults(List<no.nordicsemi.android.support.v18.scanner.ScanResult> results) {
            //for batch scan result
        }

        @Override
        public void onScanFailed(int errorCode) {
            // Scan error
        }
    };
```

philips77, 2017-07-06:
Setting in the manifest is not enough: https://developer.android.com/training/permissions/requesting.html

Jai-GAY, 2017-07-06:
i think i encounter unstable / unexpected result if i call scanner.stopScan(mScanCallback); inside the callback.

i suggest do not do that and the scan result should be inside onBatchScanResults callback and not onScanResult. try do some Log.d (); print out to diagnose.

@Override
public void onBatchScanResults(List<ScanResult> results)
{
	super.onBatchScanResults(results);
	if (results.size() > 0)
	{}
}

hope it helps.

DLeague9, 2017-07-06:
I solved it by getting permission dynamically
`private boolean checkPermission() {
        int result = this.checkSelfPermission(Manifest.permission.ACCESS_FINE_LOCATION);
        if (result == PackageManager.PERMISSION_GRANTED) {
            return true;
        } else {
            return false;
        }
    }

    private void requestPermission() {
        if (ActivityCompat.shouldShowRequestPermissionRationale(activity, Manifest.permission.ACCESS_FINE_LOCATION)) {
            Snackbar.make(findViewById(android.R.id.content), "GPS permission allows us to access location data. Please allow in App Settings for additional functionality.", Snackbar.LENGTH_LONG).show();
        } else {
            ActivityCompat.requestPermissions(activity, new String[]{Manifest.permission.ACCESS_FINE_LOCATION}, PERMISSION_REQUEST_CODE);
        }
    }`
Thanks for help.

closed 2017-07-06 by philips77

Jai-GAY, 2017-07-06:
interesting why you didn't hit [this exception](https://github.com/NordicSemiconductor/Android-Scanner-Compat-Library/issues/12) in the first place ?

philips77, 2017-07-06:
I guess it depends on the phone you are using... They log different things.