nordicsemi/Android-Scanner-Compat-Library #21: Device name is null on some conditions.

alishir, 2018-07-15:
Hi,

Sometimes the device name is null on `onBatchScanResults`, switching bluetooth on/off doesn't help, but after scanning with nRFConnect, and rescan with library the problem resolved. Is there any configuration that nRFConnect use? or is there any cache that I should clear before starting scan?

philips77, 2018-07-23:
I've heard nRF Connect is magic, but we really don't cast any spells there... :)
Btw, on Samsung S6 with some old Android release, batch scan returned all devices with the same RSSI values. An old Elephone device returned all adv packets empty.

closed 2019-01-09 by philips77

paulpv, 2019-03-22:
I see this a lot too.
BluetoothDevice.getName() often returns an unexpected null, but I have never seen ScanRecord.getDeviceName() return an unexpected null.
ScanRecord is more reliable/consistent anyway; the user can change the name of a device in the OS Bluetooth settings, and that can affect the return value of BluetoothDevice.getName(), but I have not noticed this to affect the name read by the ScanRecord.