nordicsemi/Android-Scanner-Compat-Library #23: help, Not scanning equipment

Deepblue1996, 2018-07-31:
Android 7.0 can be scanned，Android 6.0 can not scanned the device, why?

philips77, 2018-07-31:
Check the advertising packet. Avoid unnecessary 0s at the end

philips77, 2018-07-31:
Also, check the location permission.

Deepblue1996, 2018-07-31:
I have already open location permission. but i have get advertising packet is
[image]

Deepblue1996, 2018-07-31:
format Incorrect？

Deepblue1996, 2018-07-31:
nRF canned scanner

Deepblue1996, 2018-07-31:
i use your nRF test my device

Deepblue1996, 2018-07-31:
7 and 6 Bluetooth scanning protocols are different? Will filter out my equipment？

philips77, 2018-07-31:
I don't understand the issue. On what phone you can't see, and on what you can, in what app, what is the device, etc.

philips77, 2018-07-31:
No, they're the same. But some cheep phone have cheep ble controllers and they have issues.

Deepblue1996, 2018-07-31:
but android 7.0 phone Use normal, android 6.0 It's not scanned

Deepblue1996, 2018-07-31:
Curious, the system has different requirements?

Deepblue1996, 2018-08-07:
Already solved, error rate problem

closed 2018-08-07 by Deepblue1996