nordicsemi/Android-Scanner-Compat-Library #24: Android P scanner compatibility

Christophe668, 2018-08-06:
Hi, 
Has anyone ever tested the library with the latest version of Android P.
The library is working flawlessly on devices from Lollipop to Android Oreo but for some reasons, it doesn't work on my Pixel 2.
What trouble me is that the NRF connect app seems to be able to discover the device without any issue.

Thanks for the feedback !

Christophe668, 2018-08-06:
After some experiments, I found out that using `ScanFilter.Builder().setDeviceAddress` doesn't return any results on the callback.
If I remove the filter, then the callback returns the device I'm looking for.

philips77, 2018-08-06:
I'll have to check that. Thanks for the report.

JakubHyla, 2018-11-19:
I had similar problem with Huawei Mate 20 Pro. 
In my case I had to set legacy to false and phy to PHY_LE_ALL_SUPPORTED.

Based on further research it's a problem in devices with Bluetooth 5.0 (which started supporting PHY_LE_2M)

philips77, 2018-12-13:
Most lollipop+ phone support offloaded filtering, that is the low power controller allows to scan with filtering. The CPU is woken only when a matching advertising is found. There's a method on `BluetoothAdapter` *isOffloadedFilteringSupported* that says if this feature is supported. But, as we found out, this may have issues on some phones. The library allows to filter in CPU (using much more battery, but at least it's working) using the same code.

closed 2019-01-09 by philips77