nordicsemi/Android-Scanner-Compat-Library #25: Manufacturer Data / Mask seem to be ignored when filtering ScanResults

vicobz, 2018-09-04:
Hi,

I'm currently trying to set some ScanFilters when registering my BLEScanner to filter on service UUID and Manufacturer Data.

The thing is, while the ScanResults I get are properly filtered with service UUID, there is no filter applied on Manufacturer Data, resulting on inconsistent Scan Result filtering...

I tried the following: 
```
filters.add(ScanFilter.Builder()
                .setServiceUuid(ParcelUuid.fromString(serviceUUID.toString()))
                .build())

filters.add(ScanFilter.Builder()
                .setManufacturerData(manufacturerId, manufacturerData, mask)
                .build())
```
or
```
filters.add(ScanFilter.Builder()
                .setManufacturerData(manufacturerId, manufacturerData, mask)
                .setServiceUuid(ParcelUuid.fromString(serviceUUID.toString()))
                .build())
``` 

coupled with `scanner.startScan(filters, settings, callback)`

Scan Results only seem to be filtered on Service UUID.

My `manufacturerId` is an int, `manufacturerData` is a ByteArray and `mask` is a ByteArray of the same size with full `1`.

Can you please help me on this issue ?

philips77, 2019-01-11:
hi @vicobz,
Sorry for the huuuuge delay. Did you manage to solve your issue? I came back to this library after a while and I can now look into the issue.

vicobz, 2019-01-11:
Hey @philips77,

Actually I did not solve the issue and commented those lines out from my code, but still interested in it! :)

Neargye, 2019-02-08:
@vicobz maybe you try this way
```
filters.add(ScanFilter.Builder()
                .setManufacturerData(manufacturerId, manufacturerData)
                .setServiceUuid(ParcelUuid.fromString(serviceUUID.toString()))
                .build())
```
I use this way in my proj, and all look ok.

philips77, 2019-02-28:
Hi @vicobz, did the solution provided above helped?

vicobz, 2019-02-28:
Hey @philips77 , I didn't give it a try as I'm facing some critical BLE scanning issues right now. 
Could you help me a bit on how to define the appropriate mask depending on some example requirements ?

philips77, 2019-02-28:
Sure. I don't know if I'll find the time today, but you can ask here, or via email.

philips77, 2021-10-06:
I'm closing the issue due to inactivity.