nordicsemi/Android-Scanner-Compat-Library #26: The time period opens the scanning, the scanning stops, sometimes does not scan the data, after a period of time only then normal.

Deepblue1996, 2018-09-10:
private volatile boolean turnOnOff = false;

    private void startTimer() {
        timer = new Timer();
        timer.schedule(new TimerTask() {
            @Override
            public void run() {
                BluetoothLeScannerCompat scanner = BluetoothLeScannerCompat.getScanner();
                if (!turnOnOff) {
                    scanner.startScan(null, scanSettings, scanCallback);
                    Log.i("定时性", "扫描");
                } else {
                    scanner.stopScan(scanCallback);
                    Log.i("定时性", "停止");
                }
                turnOnOff = !turnOnOff;
            }
        } ,0 , 1000);
    }

Deepblue1996, 2018-09-10:
Five consecutive successes. Then fail for a while, repeat.

philips77, 2018-09-10:
That is intended. I don't know from which Android version, but when many scans are started in short period, it will not deliver results too them. You have to wait 5 or something like this seconds between scans. But even then it may happen, just after a longer time.
This can also be checked in nRF Connect of you click SCAN and STOP SCANNING quickly several times.

Deepblue1996, 2018-09-10:
Keeping a bluetooth scanner on all the time can be very power-hungry. I want to implement intermittent scanning to reduce power usage，What can be done to avoid scanning failure

philips77, 2018-09-10:
On Android 5+ you can start scanning with LOW_POWER mode. That will scan for 0.5 sec every 5 sec. This lib will also emulate the same behavior on older platforms, where it will scan and stop scanning in similar manner. Do you have the issue on an older phone?

Deepblue1996, 2018-09-10:
Scanning every five seconds is too long, and you can't get broadcast data in real time 😂

budius, 2018-09-10:
The start/stop limitation is a behavior from the Android OS, and there's nothing a library can do to avoid it. All you can do is workaround it to to avoid hitting this limitation. The exact parameters can be found on those links.
https://android-review.googlesource.com/c/platform/packages/apps/Bluetooth/+/215844/15/src/com/android/bluetooth/gatt/AppScanStats.java#144
https://web.archive.org/web/20160820074825/https://developer.android.com/preview/support.html#dp4
> We’ve changed the BLE Scanning behavior starting in DP4. We’ll prevent applications from starting and stopping scans more than 5 times in 30 seconds.

hope it helps.