nordicsemi/Android-Scanner-Compat-Library #27: BT Adapter is not turned ON

hearsilent, 2018-09-12:
I got this crash, I've check BT is on by `mBluetoothAdapter.isEnabled()`.
It's seems like sometimes `mBluetoothAdapter.isEnabled()` not the same `mBluetoothAdapter.getState() == BluetoothAdapter.STATE_ON`.

Do you have any suggestions?

Thx!

> Fatal Exception: java.lang.IllegalStateException: BT Adapter is not turned ON
       at no.nordicsemi.android.support.v18.scanner.BluetoothLeUtils.checkAdapterStateOn(BluetoothLeUtils.java:137)
       at no.nordicsemi.android.support.v18.scanner.BluetoothLeScannerImplLollipop.startScanInternal(BluetoothLeScannerImplLollipop.java:64)
       at no.nordicsemi.android.support.v18.scanner.BluetoothLeScannerCompat.startScan(BluetoothLeScannerCompat.java:123)
       at hearsilent.busplus.fragment.RouteFragment.firstTimeCheckBleDevice(RouteFragment.java:2439)
       at hearsilent.busplus.fragment.RouteFragment.access$400(RouteFragment.java:161)
       at hearsilent.busplus.fragment.RouteFragment$48$1.run(RouteFragment.java:2425)
       at android.os.Handler.handleCallback(Handler.java:739)
       at android.os.Handler.dispatchMessage(Handler.java:95)
       at android.os.Looper.loop(Looper.java:158)
       at android.app.ActivityThread.main(ActivityThread.java:7230)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1230)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1120)

closed 2018-10-24 by hearsilent

NeoSyder, 2018-11-12:
Hi, I had the same issue than @hearsilent. I can't verify if the adapter is null of its state different from
`BluetoothAdapter.STATE_ON` from `checkAdapterStateOn(BluetoothAdapter adapter)` because it was received on a phone with Raygun (= crashlytics).

I tried to invoke bluetoothAdapter from BluetoothManager instead of `BluetoothAdapter.getDefaultAdapter()` and there seems to be no difference.

philips77, 2018-11-13:
I guess what you experience here is the system trying to recover from a fault by restarting Bluetooth. Since Marshmallow, or maybe before, Android restarts Bluetooth on its own when something bad happen to it. So perhaps, when the method was started, the Bluetooth was ON, but moment later got disabled. I can't see any other reason. And I'm not sure there is any solution for that other than putting it into `try-catch` block.

NeoSyder, 2018-11-13:
Thank you for your response, this is what I thought, I was going to use a `try-catch` block.