nordicsemi/Android-Scanner-Compat-Library #29: BluetoothLeScanner on android 6.0.1 (ss s6 edge) cannot found device android >= 6.0.1

trongkhanh, 2018-10-09:
Hi all,
this code for scanning BLE device:

if (mBluetoothLEScanner == null) {
//                    create bluetoothLE scanner if it is null
                    mBluetoothLEScanner = mBluetoothAdapter.getBluetoothLeScanner();
                }
//              start scan bluetooth
                mBluetoothLEScanner.startScan(
                        null,
                        buildScanSettings(),
                        mProxyScanCallback);
 

 private ScanSettings buildScanSettings() {
        ScanSettings.Builder builder = new ScanSettings.Builder();
        builder.setScanMode(ScanSettings.SCAN_MODE_LOW_POWER);
        builder.setReportDelay(0);
        return builder.build();
    }

Setting:
<uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
    <uses-permission android:name="android.permission.BLUETOOTH" />
    <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
    <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />

I cannot found device android >= 6.0.1, but can scan other device

There are any problem?Thank you.

philips77, 2018-10-09:
Hi, Since Marshmallow you have to request Location permission (fine or coarse) in order to scan for Bluetooth LE devices. This is a Runtime permission, so you have to ask user to allow.
Check out this link: https://developer.android.com/training/permissions/requesting 

Bluetooth LE devices, like beacons, can reveal phone's location. That's why user needs to allow it. Also, on most devices the Location must be enabled in the phone.

closed 2018-10-09 by philips77

philips77, 2018-10-10:
That's a Runtime permission. You have to request it from the app, not only in the manifest. Read the link i have above.