nordicsemi/Android-Scanner-Compat-Library #32: Invoke ScanCallback in worker thread

piotr-pawlowski, 2019-01-05:
Hello,
What do you think about calling ScanCallback methods in a worker thread? It would be useful for wrapping scanner in RxJava Observable class. Right now everything is invoked in the main thread. It is simple to implement, just add a Handler param in startScan method (like in Android  SensorManager or LocationManager class).

philips77, 2019-01-07:
How about that? Note that I made also some other changes in later commits. Added annotations etc.

piotr-pawlowski, 2019-01-07:
Ok, it's better. But there is one more issue. Why `mHandler` is common for all callbacks? You support multiple scan callbacks, so you should wrap callback with its Handler. Right now every call `startScan` method override `mHandler` so it may casuse a memory leak and race condition.

philips77, 2019-01-08:
How about now?

piotr-pawlowski, 2019-01-08:
I am going to prepare PR for you, ok?

philips77, 2019-01-08:
Btw, could you take a look at #33?

piotr-pawlowski, 2019-01-08:
> Do you still find any issues in the latest impl?

There is still the issue with synchronization wrappers. But I gonna review whole library and prepare pull request.

philips77, 2019-01-31:
Fixed by #34.

closed 2019-01-31 by philips77