nordicsemi/Android-Scanner-Compat-Library #35: Scan record is always a partial byte array

kevin-objectstudio, 2019-01-09:
Not sure if I've configured this correctly, but I'm not seeing any scan results returned in my callback. It looks like the scanRecord byte[] is coming in as partial instead of complete when attempting to init a ScanRecord. When parseServiceUuid is called on line 224 inside ScanRecord.java, there is no data to parse, so I'm not seeing any returned scan results in my callback. I've implemented a different scanning service which returns the 128 bit complete data type from the same peripheral device. Am I missing some configuration settings that allows the scanner to return the completed byte[]?

I'm using the sample scanning code noted in your readme.

philips77, 2019-01-09:
On the first picture the last Record is 1 byte only (no data). 
[image]
Perhaps your device sends some weird data from time to time. On the second image the length is 17 (1 byte of type + 16 bytes of data).

philips77, 2019-01-09:
Type 6 is Incomplete List of Service 128-bit UUIDs. Type 7 (on the 2nd image) is Complete List of 128-bit UUIDs. Looks like the SoftDevice has shrunken the UUID as it could not fit the whole one. It can also shrink the Complete Local Name to Shortened Local Name automatically if it's too long.

kevin-objectstudio, 2019-01-09:
I ran through a fair amount of tests yesterday and the scan record always returns the same data. So if it's returning the incomplete list, is there a way to wait for the remainder of the data to come through? This is where the service uuid is being checked. Since it comes back as incomplete, when ScanFilter.java starts matching the service uuid I passed into my filter when I started the scan, with what is returned in the scan result, the matches method inside ScanFilter.java on line 280 always returns false. Which means I never see any results passed to my onScanResult callback.

closed 2019-01-09 by kevin-objectstudio