nordicsemi/Android-Scanner-Compat-Library #41: Scanning multiple iBeacon devices

Eroshkevich, 2019-01-22:
Hello!
We are trying to use AndroidScannerCompatLibrary in our Android application to scan BLE devices and measure its RSSI level. BLE devices is mounted on ceiling with distance approximately 1m.
There is implementation of StartScan() action (its uses Xamarin Binding Library from Nordic AndroidScannerCompatLibrary .aar file):
```
var scanner = BluetoothLeScannerCompat.Scanner;
            ScanSettings settings = new ScanSettings.Builder()
               .SetLegacy(false)
               .SetScanMode(ScanSettings.ScanModeLowLatency)
               .SetMatchMode(ScanSettings.MatchModeAggressive)
               .SetNumOfMatches(ScanSettings.MatchNumMaxAdvertisement) использую
               .SetReportDelay(0)
               .SetUseHardwareBatchingIfSupported(false)
               .Build();
           List< ScanFilter > filters = new List<ScanFilter>();
scanner.StartScan(filters, settings, _signalsScanCallback);
```
_signalsScanCallback store signals in package. After some period of time application send package to server storage.
And this is our stored RSSI level data set (from two iBeacon devices):
[image]
It seems like we have data from “one-device-per-period-of-time”.
Should we StartScan() for each device separately with list of ScanFilters, or it’s ok to use one instance of Scanner for all of it? We try to compare our results with same measures from “nRf connect”, and it’s differing. Can someone explain what are we doing wrong?

philips77, 2019-01-23:
It should work with a single scan and no filter (null or empty list) or a filter that matches the Manufacturer Data of both iBeacons. 

`.SetUseHardwareBatchingIfSupported(false)` may be removed, as you have the report delay set to 0. But it's ignored now, so doesn't really matter.

In nRF Connect the main RSSI graph displays average RSSI values over at most 10 samples from last 4 seconds. It is refreshed every 0.5 second. Even if a device advertises every second or two there will be a line shown and its value may change, as some older samples may now not be included in the averaging. That's why you may see a difference. 

There is also a way to show exact RSSI values in a Timeline in nRF Connect. Select those 2 devices by long tapping a first one and then short tapping another one and select Timeline icon in the top menu. Could you post an image how the graph looks like in there?

What would be an expected result?

Eroshkevich, 2019-01-24:
Thank you for your answer, philips))

> There is also a way to show exact RSSI values in a Timeline in nRF Connect. Select those 2 devices by long tapping a first one and then short tapping another one and select Timeline icon in the top menu. Could you post an image how the graph looks like in there?

Ok, thanks for this feature. Sorry that there is no way to download log file with this values and timestamps ;-)
We compare our and "nRf" graphs and came to the conclusion that it's differ because of running not on the same Android device :-8

Can you explain some more about averaging algorythm?

> In nRF Connect the main RSSI graph displays average RSSI values over at most 10 samples from last 4 seconds. It is refreshed every 0.5 second. Even if a device advertises every second or two there [...]

My understanding: Compute average of last 10 values within last 4s every 0.5s. Is it correct?

philips77, 2019-01-24:
You understanding is correct. More precisely, if a device advertises every second only 4 packets will be used to compute average. If every 20 ms -> 10 last packets (from last 200 ms), etc. If a device stopped advertising it will be shown for the next 4 seconds and the line will smother as less and less packets will be averaged, until it will disappear after that time.

There is a task: https://github.com/NordicSemiconductor/Android-nRF-Connect/issues/78 to make this values exportable. But resolution time is unknown.

closed 2019-01-31 by philips77