nordicsemi/Android-Scanner-Compat-Library #43: Connect hangs on some devices

nicrgren, 2019-01-31:
Hi,

Using your scanner compat library I've noticed that the scanner sometimes discovers devices where the record.device.name is null. Whenever the scanner identifies one of your boards like this I cannot connect to it - BleManager.connect(device) simply times out without providing any callbacks.

To resolve this problem I start nRF Connect which then causes the scanner to correctly find the friendly name of the devices as well as allowing me to connect to them.

It feels like I am missing something with the scanner, you seem to somehow reset the adapter. How is this done?

Thanks for some great products!

philips77, 2019-01-31:
Hi, did you manage to solve the issue?

nicrgren, 2019-02-01:
Hi Philips, Sorry for not replying.

It seems removing `setReportDelay` as you suggested resolved the issue. Thank you for you assistance :)

Sorry if I created this issue in the wrong repository!

closed 2019-02-01 by nicrgren