nordicsemi/Android-Scanner-Compat-Library #45: [1.3.0][Android < Oreo] Scanning stops when app is swiped out of the "task-manager"

fastrde, 2019-02-11:
The new pendingIndent Solution works great on Android >= Oreo.
I wrote a foreground service that handles the BLE stuff independently from the actual app. This service is the only place where startScan/stopScan is called.

When I call `scanner.startScan(filters, settings, mService, pendingIntent);` the scanning stops when i swipe the app out of the android "task-manager" (The foreground-service is still running). When I call the scanner with `scanner.startScan(filters, settings, scanCallback);` the scanning didn't stop. 

I think this is the normal background task behavior when the task is started from an activity in the app. i wonder why the background task get destroyed when I start the scan in the foreground service.

fastrde, 2019-02-12:
Thank you for your quick response. The removal of `stopSelf()` does the trick. The service doesn't get killed. I'll check next if the service respond to my BLE Device after hours in the background and let you know.

fastrde, 2019-02-13:
When the phone [Lollipop < affected Version < Oreo] enters doze mode announcing devices are not found by the scanner. I think there is nothing that can be done about it (except asking the user for whitelisting), even foreground apps and services are restricted in doze mode.

See https://developer.android.com/training/monitoring-device-state/doze-standby

philips77, 2019-02-15:
I will address your findings in the new release.

philips77, 2019-02-15:
`stopSelf()` removed in 1.3.1.

closed 2019-02-15 by philips77