nordicsemi/Android-Scanner-Compat-Library #47: Pending Intent not send Device data

Rakesh4a7, 2019-02-28:
Pending Intent callback doesn't provides the Scan result data. Can you provide an example for the same if it's already available?

philips77, 2019-02-28:
Could you post the relevant code here? 
The scan results you receive should be in the `Intent` you get in your `BroadcastReceiver`. You have to register the receiver in the `Context`, not using `LocalBroadcastReceiver`. Remember, that you should use the compat keys to extract data:
https://github.com/NordicSemiconductor/Android-Scanner-Compat-Library/blob/569982877c77c005649f728809d3199faf2080af/scanner/src/main/java/no/nordicsemi/android/support/v18/scanner/BluetoothLeScannerCompat.java#L68-L88

Rakesh4a7, 2019-02-28:
Hi [philips77](/philips77) 

Thanks for quick response. 

Now I'm able to get the scan results. But when my app is killed i'm not receiving the Broadcast receiver.  can you check it. Below is my code.

` Intent intent = new Intent(this, MyReceiver.class); // explicite intent
        intent.setAction("com.example.ACTION_FOUND");
        PendingIntent pendingIntent = PendingIntent.getBroadcast(this, 1, intent, PendingIntent.FLAG_UPDATE_CURRENT);

        BluetoothLeScannerCompat scanner = BluetoothLeScannerCompat.getScanner();
        ScanSettings settings = new ScanSettings.Builder()
                .setLegacy(false)
                .setScanMode(ScanSettings.SCAN_MODE_LOW_LATENCY)
                .setReportDelay(1000)
                .setPhy(ScanSettings.PHY_LE_ALL_SUPPORTED)
                .build();
        List<ScanFilter> filters = new ArrayList<>();
        ScanFilter filter = new ScanFilter.Builder().setDeviceAddress("00:0B:57:47:D5:EA").build();
        filters.add(filter);
        scanner.startScan(filters, settings, this, pendingIntent);`

`public class MyReceiver extends BroadcastReceiver {
    @Override
    public void onReceive(Context context, Intent intent) {
        if(intent.hasExtra(BluetoothLeScanner.EXTRA_LIST_SCAN_RESULT)){
            ArrayList<ScanResult> results = intent.getParcelableArrayListExtra(BluetoothLeScanner.EXTRA_LIST_SCAN_RESULT);

            for (final ScanResult result : results) {
                Log.e("onScanResult","onScanResult : " + result.getDevice().getName() + " " + result.getDevice().getAddress());
                Toast.makeText(context, result.getDevice().getName() + " " + result.getDevice().getAddress(), Toast.LENGTH_LONG).show();
            }
        }

      /*  StringBuilder sb = new StringBuilder();
        sb.append("Action: " + intent.getAction() + "\n");
        sb.append("URI: " + intent.getDataString() + "\n");
        String log = sb.toString();
        Log.d("MyReceiver", log);
        Toast.makeText(context, log, Toast.LENGTH_LONG).show();*/
    }
}`

`<manifest xmlns:android="http://schemas.android.com/apk/res/android"
    package="com.example.bluetoothdemo">

    <uses-permission android:name="android.permission.BLUETOOTH" />
    <uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
    <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />

    <uses-feature
        android:name="android.hardware.bluetooth_le"
        android:required="true" />

    <application
        android:allowBackup="true"
        android:icon="@mipmap/ic_launcher"
        android:label="@string/app_name"
        android:roundIcon="@mipmap/ic_launcher_round"
        android:supportsRtl="true"
        android:theme="@style/AppTheme">
        <activity android:name=".MainActivity">
            <intent-filter>
                <action android:name="android.intent.action.MAIN" />

                <category android:name="android.intent.category.LAUNCHER" />
            </intent-filter>
        </activity>

        <receiver android:name="MyReceiver">
            <intent-filter>
                <action android:name="com.example.ACTION_FOUND">
                </action>
            </intent-filter>
        </receiver>
    </application>

</manifest>`

philips77, 2019-02-28:
What Android API level are you testing?

philips77, 2019-02-28:
Btw, in teh `BroadcastReceiver` you should use `BluetoothLeScannerCompat.EXTRA_LIST_SCAN_RESULT` instead of  `BluetoothLeScanner.EXTRA_LIST_SCAN_RESULT`.

Rakesh4a7, 2019-02-28:
I'm testing on API 28

philips77, 2019-02-28:
Emulating `PendingIntent` scanning is done using a `Service`. If your app gets killed, also does the service. The new version should survive removing task from Recents, but forcing the app to be killed will also stop the service and there is nothing that can be done about that.

paulpv, 2019-03-11:
@philips77 If this is intended and expected then I suggest you remove/edit/ammend the following comment from the readme.md:
> This will also work after your application has been killed (the receiver must be added in AndroidManifest and the PendingIntent must be created with an explicit Intent).

philips77, 2019-03-12:
It depends on the system version. On Oreo+ the result will be delivered despite app being killed. On previous versions is all about preserving the Scanner Service from killing. Removing a task from Recents should not affect it, services are not killed like on Oreo, so may live hidden in bg, but Force stopping the app will kill the service.