nordicsemi/Android-Scanner-Compat-Library #48: I can`t stop scan,After I call the stop scan method, the callback is still executed.

funaihui, 2019-03-01:
```java
 override fun onBatchScanResults(results: List<ScanResult>?) {
            // This callback will be called only if the report delay set above is greater then 0.

            // If the packet has been obtained while Location was disabled, mark Location as not required
            if (Utils.isLocationRequired(application) && !Utils.isLocationEnabled(application))
                Utils.markLocationNotRequired(application)

            var atLeastOneMatchedFilter = false
            for (result in results!!)
                atLeastOneMatchedFilter = mDevicesData.deviceDiscovered(result) || atLeastOneMatchedFilter
            if (atLeastOneMatchedFilter) {
                mDevicesData.applyFilter()
                mScannerState.recordFound()
            }
            stopScan()
        }

fun stopScan(){
        val scanner = BluetoothLeScannerCompat.getScanner()
        scanner.stopScan(scanCallback)
    }

```

btoconnor, 2019-10-24:
Any update on this?  I'm trying something similar (just to see things working) and I'm seeing the same behavior.  My callback keeps getting called after I run `stopScan` on a scanner object.

(I'm sure I'm doing something wrong, but I'm not entirely sure how to fix the problem)

liu1700, 2021-02-17:
I think it may be caused by you are using the different ` scanCallback` instance
for 
`scanner.startScan(filters, settings, scanCallback);` and  `scanner.stopScan(scanCallback);`

using the same `scanCallback` instance fix the issue in my case.

closed 2021-10-06 by philips77