nordicsemi/Android-Scanner-Compat-Library #53: Add a scanning operation timeout

XabierGoros, 2019-03-28:
It would be good to have a scanning timeout to set a maximum of the process if desired. Something like this in the API would be nice:

```java
@NonNull
public Builder setDuration(final long durationMillis) {
  if (durationMillis < 0) {
    throw new IllegalArgumentException("durationMillis must be > 0");
  }
  this.durationMillis = durationMillis;
  return this;
}
```

and start a `Handler`with that specific `durationMillis` until it is reached to stop the scanning process. I guess that would require a new `onStop` method in the `ScanCallback` abstract class.

What do you think guys? Is it something the library could handle or it's just better to manage ourselves those timeouts?

philips77, 2021-08-31:
The main purpose of this library is to bring the new API (or emulate it) to older Android versions, not extend it. The feature you asked for would be the latter type. I agree it can be useful, but it can easily be done in the app.
Also, sorry for not replying to you in more than 2 years!

closed 2021-08-31 by philips77

LuoPeiQin, 2022-09-15:
This is very useful to me, is there any development plan for this feature now?