nordicsemi/Android-Scanner-Compat-Library #54: Unable to scan devices

satishab, 2019-04-01:
Hi,

I have used the Sample provided in the description, without any filter. 
But still I am not getting callbacks. I have also referred to nrf-blinky example which uses this library. 
My Scan code is exactly same as that provided in except filter part https://github.com/NordicSemiconductor/Android-nRF-Blinky.  But still no callbacks are received. 

Phone Model - Oneplus 3.
Note - I am trying to search and connect a non nordic BLE device. will that have any impact on this?
Please help I am stuck on this :(
Thanks,
sat

satishab, 2019-04-01:
Also it will be great if you could give me working example simple one is also fine just for scanning also is fine

satishab, 2019-04-01:
Android version 8.0

philips77, 2019-04-01:
Hello, did you request and grant Location permission to your app? And do you have the Location enabled in the Quick Settings in Android?

satishab, 2019-04-01:
Yes

satishab, 2019-04-01:
Do you have simple also fine but scanning example?
Do I am missing anything ??

philips77, 2019-04-01:
Please, paste some of your code here.

core2root, 2019-05-02:
Have same problem, i requested location permission, call for startScan() but no even one callback triggered.
*Note that i successfully able to use the androids stack scanning and receive results.
Samsung s8 android 9
Xiaomi mi 6 android 8

core2root, 2019-05-02:
> Have same problem, i requested location permission, call for startScan() but no even one callback triggered.
> *Note that i successfully able to use the androids stack scanning and receive results.
> Samsung s8 android 9
> Xiaomi mi 6 android 8

After some checks i found that by default the library is set to return results in batches so to be able receive results need to override the `onBatchScanResults` or call `.setUseHardwareBatchingIfSupported(false)` on the ScanSettings

closed 2019-05-03 by philips77