nordicsemi/Android-Scanner-Compat-Library #56: Scan result name is null

core2root, 2019-05-02:
Hi, i have some weird behavior with ble scanning.
I implemented app that perform scanning and connection to ble devices with help of android scanner, the problem is i have some devices (for example samsung s8 android version 9) that the scan result not contains the device name, i checked from other android device i do see the name.
I downloaded the NrfConnect app from google play store and it seems that this app show the device name at 80 % of the time in the problematic device (samsung s8) , so i downloaded this library to see what is the 'magic' that helps to receive the name, but after building simple app that prints the name to log i seed that even with current library the names are null always. 
The question is what can lead to this problem ? Do you use current library in the NrfConnect app ?

```
}
....
if(Build.VERSION.SDK_INT >= Build.VERSION_CODES.M){
      if(checkSelfPermission(Manifest.permission.ACCESS_FINE_LOCATION) != PackageManager.PERMISSION_GRANTED){
        requestPermissions(new String[]{Manifest.permission.ACCESS_FINE_LOCATION}, 23);
      }
    }

    final BluetoothLeScannerCompat scanner = BluetoothLeScannerCompat.getScanner();

    final ScanSettings settings = new ScanSettings.Builder()
            .setScanMode(ScanSettings.SCAN_MODE_LOW_LATENCY)
            .setUseHardwareBatchingIfSupported(false)
            //.setReportDelay(5000)
            .build();
    final List<ScanFilter> filters = new ArrayList<>();
    filters.add(new ScanFilter.Builder().setServiceUuid(new ParcelUuid(UUID.fromString("XXXXXX-XXXX-0BAC-5241-XXXXXX"))).build());

    startScanBtn.setOnClickListener(new View.OnClickListener() {
      @Override
      public void onClick(View v) {
        scanner.startScan(filters, settings,listener);
      }
    });

    stopScanBtn.setOnClickListener(new View.OnClickListener() {
      @Override
      public void onClick(View v) {
        scanner.stopScan(listener);
      }
    });
  }

  private ScanCallback listener = new ScanCallback() {
    @Override
    public void onScanResult(int callbackType, ScanResult result) {
      super.onScanResult(callbackType, result);
      String name = result.getDevice().getName();
      Log.d("OnMyScanResult", "DeviceName: "+name);
    }

    @Override
    public void onScanFailed(int errorCode) {
      super.onScanFailed(errorCode);
      Log.d("ScanResult", "Scan failed code: "+errorCode);
    }

    @Override
    public void onBatchScanResults(@NonNull List<ScanResult> results) {
      super.onBatchScanResults(results);

      for(ScanResult result : results){
        String name = result.getDevice().getName();
        Log.d("OnMyScanResult", "DeviceName: "+name);
      }
    }
  };`
```

philips77, 2019-05-03:
nRF Connect is not using the scanner Compat Library. It actually allows you to switch between new and old scanning API in settings.
But in the app I'm not taking the name from BluetoothDevice, instead i parse the raw data. In the scanner library you may get the name from `result.getRecord().getName()`. Such name is parsed from the received data, not taken from phone's cache. The name in BluetoothDevice object gets overridden by value from Device Name characteristic after connection and may be cashed and not updated.

core2root, 2019-05-03:
This is very weird, i do tried to take the bytes from scan record but only 0-30 indexes had a value the rest 31-63 was zeroes, and from debugging i saw that the name is sit in 31-42 indexes .

philips77, 2019-05-03:
The raw bytes contain both advertising data and scan response data, if such is present. Reach of them is 0-31 bytes going. The rest of the array indeed is filled with zeros. Perhaps your phone indirectly puts scan response to the raw data and then fails to parse it?

core2root, 2019-05-03:
I really don't know why this happens, because the bluetooth device for sure send the scan response (other phones can see it) , the real question is why the NrfConnect app do see 80% of our ble devices ( we develop smart locks with ble) , where i make the check i have about 200 - 400 ble devices, and my custom bytes parsing show that the are no name contained in the data.

philips77, 2019-05-04:
If you're parsing the raw data manually, did you check Shortened Local Name field? It also may contain the name.

core2root, 2019-05-04:
I checked, the short name not included in the data.

khan-07, 2019-07-25:
Hi,

I am facing the same problem . I can not get names from either `getDevice()` or 'getScanRecord()'. I am working on android 9 as well.  I get null for 99% of the times but sometimes i get a device name.

core2root, 2019-07-25:
Hi, @khan-07  i think i found the root cause for this problem and even the solution (not tried yet)
Look at the last answer [here](https://github.com/don/cordova-plugin-ble-central/issues/665)

closed 2021-10-06 by philips77