nordicsemi/Android-Scanner-Compat-Library #57: ManufacturerSpecificData is not working properly

Ruifen9, 2019-06-27:
I have a low-power Bluetooth device that advertisement data like this: 
[
    02 01 05 08 09 61 6e 74 68 65 72 6f 
    09 ff 11 02 11 02 d9 21 a3 38 
    1e ff 11 02 11 02 d9 21 a3 38 21 43 01 d9 00 10 19 44 45 4c 20 68 73 65 4d 20 47 53 54 42 
     4a 00 00 00 00 00 00 00 00 00
]
// result.getScanRecord().getManufacturerSpecificData() 
//or
//result.getScanRecord().getManufacturerSpecificData(0x0211)

The ManufacturerSpecificData I read sometimes is like this：
[ 11 02 d9 21 a3 38 ]  (When the signal is weak, more is like this)
, sometimes it is like this:
[
     11 02 d9 21 a3 38 21 43 01 d9 00 10 19 44 45 4c 20 68 73 65 4d 20 47 53 54 42 
     4a 00 00 00 00 00 00 00 00 00
]
And I want to get all the data, but I didn't find the relevant method.
I hope I can get your reply, thank you.

philips77, 2019-06-27:
With this: https://developer.android.com/reference/android/bluetooth/le/ScanRecord.html#getBytes() method you may get all the data received, not only Manufacturer Data. I think, if the data are truncated, the only solution is to ignore such packet and wait for another.

Ruifen9, 2019-07-05:
Thank you very much for your reply. This method will still get incomplete data, and most of them come from devices with weak signals.

closed 2019-07-05 by Ruifen9