nordicsemi/Android-Scanner-Compat-Library #60: Disabling background scan when device is locked

schlachtzeuger, 2019-07-23:
Hi there,

I tried the background scanning and it works just perfectly except for devices running Android 7 or lower where this background scanning behavior is emulated as the battery drains really quickly.

On devices with O or P, I saw that the scanning is normally suspended by the OS some time after the device was locked. Is it possibly to do this too or (even better) to just throttle the background scanning then?

Regards

Max

ndrjt, 2019-07-31:
Hi @schlachtzeuger,
I am facing the same issue, even I have used pendingintent for background scan for a while but it scanning for around 30 mins. is there any way to scan continuously for hours and hours? 

Thanks.

philips77, 2019-07-31:
I think the design is to prevent the battery from dying. You may have a Task in TaskManager that will start and stop scanning every now and then if you really need to.

ndrjt, 2019-07-31:
Yeah, what if I need to scan continuously? And how we can achieve it by library?

philips77, 2019-08-01:
The goal of the library is to mimic the behavior of the latest Android version on older phones. I'm rather thinking of focusing on saving battery for pre-Oreo phones, than on allowing long lasting continuous scan for newer.

indrajit9, 2019-08-01:
Thank you @philips77

schlachtzeuger, 2019-08-01:
@ndrjt : After reading my request I'm not sure if I explained my issue comprehensible: My issue is not that it stops after some time but that it literally kills the battery on older Android versions.

philips77, 2019-08-02:
Well, that's understandable. Older versions don't have offloaded scanning, they have to run the service and even filter devices using CPU. I would even say, that for long lasting scanning the minimum Android version should be 8, where the option with Pending Intent was added. On older phones it is not possible to have all the features: background scanning, quick responses, filtering, and good battery lifetime.

schlachtzeuger, 2019-09-03:
That's exactly why I was asking if we could throttle background scanning on pre-Oreo devices a bit. Thus we could have some sort of background scanning, maybe with a higher delay but without killing the battery.

philips77, 2021-06-04:
There is this method to trim scan and rest intervals:
https://github.com/NordicSemiconductor/Android-Scanner-Compat-Library/blob/e56d7f134600ee10f611660d53ff3040c08e463e/scanner/src/main/java/no/nordicsemi/android/support/v18/scanner/ScanSettings.java#L575-L593
I'm closing the issue due to inactivity. Sorry that it took me so long to reply.

closed 2021-06-04 by philips77