nordicsemi/Android-Scanner-Compat-Library #65: Android Lollipop "blocking" some devices on scan results

Bertoleti, 2019-09-14:
Hi,
I have one problem in scan devices on Android Lollipop (5.1.1).

I have one device advertising constantly and sometimes the library don't return this device in advertisement results.
I was investigating the problem and I noted that when the lib don't return the advertisement anyone BLE app returns that advertisement.
It's like the Android has blocked that "device" and anyone app can't identify.

On newer Android versions I don't have this problem.

Someone knows what's happening?

PS: There is a fun fact that looking the LOG, I can see de BLE Address with and error, but I don't found any explanation about that.

09-15 15:09:01.268 4379-4579/? E/bt-btm: new address: 55:44:33:22:11:00

where my BLE Address is 55:44:33:22:11 but the result don't appear in the return scanned devices on the lib.

I have tried to filter by BLE Address, services, manufacturer data, etc etc, but I have the same problem. 

Thanks in advanced.

Bertoleti, 2019-09-14:
The problem above occurs when I'm using the LOW_LATTENCY_MODE, when I change to BALANCED_MODE, I start to have results, but, ONLY when I start an scan process by other APP.

When this other app stop scan, the lib don't "see" the advertising any more.

philips77, 2021-06-04:
Hello,
First of all, sorry for the 1.5 years delay. I was busy with other tasks and missed your issue. Did you manage to solve it? Can we close the issue?