nordicsemi/Android-Scanner-Compat-Library #66: Extra in callback intent

schlachtzeuger, 2019-09-17:
Hi there,

as in the example code, I try to add an extra to a callback intent, but this leads to some interesting results:

First, only the added extra shows up in the intent, meaning that BluetoothLeScannerCompat.EXTRA_LIST_SCAN_RESULT was removed.

But also I get massive stacktraces in logcat:

`
2019-09-17 10:33:02.790 1365-3432/? E/Parcel: Class not found when unmarshalling: no.nordicsemi.android.support.v18.scanner.ScanResult
    java.lang.ClassNotFoundException: no.nordicsemi.android.support.v18.scanner.ScanResult
        at java.lang.Class.classForName(Native Method)
        at java.lang.Class.forName(Class.java:453)
        at android.os.Parcel.readParcelableCreator(Parcel.java:2811)
        at android.os.Parcel.readParcelable(Parcel.java:2765)
        at android.os.Parcel.readValue(Parcel.java:2668)
        at android.os.Parcel.readListInternal(Parcel.java:3098)
        at android.os.Parcel.readArrayList(Parcel.java:2319)
        at android.os.Parcel.readValue(Parcel.java:2689)
        at android.os.Parcel.readArrayMapInternal(Parcel.java:3037)
        at android.os.BaseBundle.initializeFromParcelLocked(BaseBundle.java:288)
        at android.os.BaseBundle.unparcel(BaseBundle.java:232)
        at android.os.Bundle.putAll(Bundle.java:288)
        at android.content.Intent.fillIn(Intent.java:9433)
        at com.android.server.am.PendingIntentRecord.sendInner(PendingIntentRecord.java:269)
        at com.android.server.am.PendingIntentRecord.sendWithResult(PendingIntentRecord.java:245)
        at com.android.server.am.ActivityManagerService.sendIntentSender(ActivityManagerService.java:10000)
        at android.app.IActivityManager$Stub.onTransact$sendIntentSender$(IActivityManager.java:11540)
        at android.app.IActivityManager$Stub.onTransact(IActivityManager.java:3183)
        at com.android.server.am.ActivityManagerService.onTransact(ActivityManagerService.java:3829)
        at android.os.Binder.execTransact(Binder.java:752)
     Caused by: java.lang.ClassNotFoundException: no.nordicsemi.android.support.v18.scanner.ScanResult
        at java.lang.Class.classForName(Native Method)
        at java.lang.BootClassLoader.findClass(ClassLoader.java:1346)
        at java.lang.BootClassLoader.loadClass(ClassLoader.java:1406)
        at java.lang.ClassLoader.loadClass(ClassLoader.java:312)
        at java.lang.Class.classForName(Native Method) 
        at java.lang.Class.forName(Class.java:453) 
        at android.os.Parcel.readParcelableCreator(Parcel.java:2811) 
        at android.os.Parcel.readParcelable(Parcel.java:2765) 
        at android.os.Parcel.readValue(Parcel.java:2668) 
        at android.os.Parcel.readListInternal(Parcel.java:3098) 
        at android.os.Parcel.readArrayList(Parcel.java:2319) 
        at android.os.Parcel.readValue(Parcel.java:2689) 
        at android.os.Parcel.readArrayMapInternal(Parcel.java:3037) 
        at android.os.BaseBundle.initializeFromParcelLocked(BaseBundle.java:288) 
        at android.os.BaseBundle.unparcel(BaseBundle.java:232) 
        at android.os.Bundle.putAll(Bundle.java:288) 
        at android.content.Intent.fillIn(Intent.java:9433) 
        at com.android.server.am.PendingIntentRecord.sendInner(PendingIntentRecord.java:269) 
        at com.android.server.am.PendingIntentRecord.sendWithResult(PendingIntentRecord.java:245) 
        at com.android.server.am.ActivityManagerService.sendIntentSender(ActivityManagerService.java:10000) 
        at android.app.IActivityManager$Stub.onTransact$sendIntentSender$(IActivityManager.java:11540) 
        at android.app.IActivityManager$Stub.onTransact(IActivityManager.java:3183) 
        at com.android.server.am.ActivityManagerService.onTransact(ActivityManagerService.java:3829) 
        at android.os.Binder.execTransact(Binder.java:752) 
     Caused by: java.lang.NoClassDefFoundError: Class not found using the boot class loader; no stack trace available
2019-09-17 10:33:02.790 1365-3432/? W/Bundle: Failed to parse Bundle, but defusing quietly
    android.os.BadParcelableException: ClassNotFoundException when unmarshalling: no.nordicsemi.android.support.v18.scanner.ScanResult
        at android.os.Parcel.readParcelableCreator(Parcel.java:2839)
        at android.os.Parcel.readParcelable(Parcel.java:2765)
        at android.os.Parcel.readValue(Parcel.java:2668)
        at android.os.Parcel.readListInternal(Parcel.java:3098)
        at android.os.Parcel.readArrayList(Parcel.java:2319)
        at android.os.Parcel.readValue(Parcel.java:2689)
        at android.os.Parcel.readArrayMapInternal(Parcel.java:3037)
        at android.os.BaseBundle.initializeFromParcelLocked(BaseBundle.java:288)
        at android.os.BaseBundle.unparcel(BaseBundle.java:232)
        at android.os.Bundle.putAll(Bundle.java:288)
        at android.content.Intent.fillIn(Intent.java:9433)
        at com.android.server.am.PendingIntentRecord.sendInner(PendingIntentRecord.java:269)
        at com.android.server.am.PendingIntentRecord.sendWithResult(PendingIntentRecord.java:245)
        at com.android.server.am.ActivityManagerService.sendIntentSender(ActivityManagerService.java:10000)
        at android.app.IActivityManager$Stub.onTransact$sendIntentSender$(IActivityManager.java:11540)
        at android.app.IActivityManager$Stub.onTransact(IActivityManager.java:3183)
        at com.android.server.am.ActivityManagerService.onTransact(ActivityManagerService.java:3829)
        at android.os.Binder.execTransact(Binder.java:752)
`

I was testing my code on a OnePlus 5T running Android 9, scanner lib is at 1.4.3.

Does anyone know what's going on there?

Cheers,

Max

philips77, 2021-06-04:
Hi,
First of all, sorry for 1.5 years delay. I was busy with other tasks and missed your issue. Did you manage to solve the problem? Can we close the issue?

philips77, 2021-09-29:
Closing due to inactivity.

closed 2021-09-29 by philips77