nordicsemi/Android-Scanner-Compat-Library #67: In Android 9 and 10 if use filter peer not found on scan process

LebetsVI, 2019-09-30:
Hi.
Using this lib, I came across an unpleasant thing: if I set  filter like as:

_final List<ScanFilter> filters = new ArrayList<>();
filters.add(new ScanFilter.Builder().setDeviceName(KNOWN_DEV_NAME).build());
scanner.startScan(filters, settings, scanCallback);_

then scanner do not return result and stop by timeout. 

Any settings for "setScanMode" do not give a result.

When don`t use filter: all ok.

philips77, 2021-06-04:
Hi,
Sorry for 1.5 year delay. Did you manage to solve the issue?
I know that offloaded filtering not always works, that's why we have
https://github.com/NordicSemiconductor/Android-Scanner-Compat-Library/blob/e56d7f134600ee10f611660d53ff3040c08e463e/scanner/src/main/java/no/nordicsemi/android/support/v18/scanner/ScanSettings.java#L499-L514 flag to emulate filtering in the code (which uses much more battery).

LebetsVI, 2021-06-05:
Hi.
I set "ScanSettings" with "setUseHardwareFilteringIfSupported(**false**)", don`t use lib-embedded filtering and use our custom filter in advertising packets.

closed 2021-09-29 by philips77