nordicsemi/Android-Scanner-Compat-Library #68: Scanning with PendingIntent after connection was closed

schlachtzeuger, 2019-10-08:
Hi there,

I'm currently struggling with a very strange behavior that I can't explain and I hope that anyone of you has an idea. I use this lib in version 1.4.3 on a Samsung S10 (Android 9) and a Pixel 3a (Android 10).

When I found a BLE device, I stop scanning, connect to the device, do some stuff and then close the connection. After that, I start a new scan. Now when using the preferred way with a PendingIntent, it takes around 9.5 seconds until I get the new scan results. But when I simply use a ScanCallback, the new results come in immediately.

Can anyone confirm this experience? Or any idea what probably went wrong?

Thanks in advance!

Max

philips77, 2019-10-08:
That is expected on Android 8+. Scanning with pending intent is designed to use very low power so the BLE controller on your phone may scan in power saving mode. This will typically take more time than high priority scanning.

schlachtzeuger, 2019-10-08:
Sure, I get the point. But if I just start and stop scanning, I get results immediately. Only if I connected before, it's delayed. What I saw is that the ACL_DISCONNECTED action comes exactly at the point where the scan gets results again, so I assume the connection probably wasn't really disconnected before.

After disconnection, I start the scan and only 80ms later, I get the first entries on logcat telling
`10-08 16:05:55.330  6016  6096 D BtGatt.GattService: onScanResult to scannerId: 8- eventType=0x10, addressType=1, address=CD:B, primaryPhy=1, secondaryPhy=0, advertisingSid=0xff, txPower=127, rssi=-90, periodicAdvInt=0x0
10-08 16:05:55.331  2478  2478 D BluetoothAdapter: STATE_ON
10-08 16:05:55.333  6016  6096 D ScanRecord: Not a Multi Manu data
10-08 16:05:55.333  6016  6096 D ScanRecord: Not a Multi Manu data`

so it seems that the scanning process itself is running. Some time after that, these messages are accompanied with the callbacks to my receiver.

I put together a small example if someone wants to try out:
[ScanExample.zip](https://github.com/NordicSemiconductor/Android-Scanner-Compat-Library/files/3702823/ScanExample.zip)

schlachtzeuger, 2019-10-22:
It seems to be a general Android issue. FYI, I filed a bug at Google (https://issuetracker.google.com/issues/142404995) so I think this issue can be closed.

closed 2019-10-22 by schlachtzeuger