nordicsemi/Android-Scanner-Compat-Library #69: onBatchScanResults() returns a device even after it stops advertising

acutetech, 2019-10-08:
Three related questions:

Q1: I set .setReportDelay(5000) then get calls to onBatchScanResults() every 5s, initially with no devices. I switch on a device and onBatchScanResults() reports that device. However, it continues to report the same device even when I power it off. This continues until I call stopScan(). I am not connecting to the device. Is this expected behaviour, and how can I stop it (e.g. clear the device from some cache somewhere.)

Q2: If my settings do not include .setReportDelay(5000) then I get calls to onScanResult() instead of onBatchScanResults(). Is that the expected behaviour? (Here I have to stop scanning or I get a flood of reports of the same device.)

Q3: I see mention of not turning off scanning inside the ScanCallback. Why, and where should it be turned off (when I want to conn

acutetech, 2019-10-09:
Hmmm. I am not getting this issue today. I wonder if I could have had multiple scanners operating as a result of repeated downloads of the app to my device during debugging? Or something... Any tips on scanner lifecycles?

Any insight appreciated, as well as answers to the questions anyway...

acutetech, 2019-10-09:
Sorry, back with the same problem. The same image on 3 different Android devices. Two continue to report finding a BLE device after it has been switched off. One stops reporting the BLE device when I power it off.

philips77, 2019-10-22:
Hello, 
Sorry for the late response. 
Regarding Q2: yes, this is expected behaviour. As one of the parameters you should also get a `callbackType`, which may be one of callback types given here: https://developer.android.com/reference/android/bluetooth/le/ScanSettings.html
However, callback types first match and match lost don't work on many phones. I didn't test it recently, but on the older phones I never manage to make it work.
Regarding Q1: we usually don't scan with delay in our apps, as we need the results as soon as possible, so we didn't test this feature well. As you wrote, some phones report a lost device, some don't. Perhaps some devices are reported with `callbackType` match lost, but in this callback you actually don't get the callback type, so you never know whether the device is still advertising or not. Is there a difference in packet content? E.g. is the advertising data empty after you turn off the device?
Regarding Q3: we do turn off scanning inside `ScanCallback`. I don't know why not. In my opinion it's a good place, like any other.

Btw, did you try setting different `ScanSettings` (https://developer.android.com/reference/android/bluetooth/le/ScanSettings.Builder.html)? For example the callback type, num of matches and match mode? Do you see any differences in what you get?

philips77, 2021-06-04:
I'm closing it due to inactivity.

closed 2021-06-04 by philips77