nordicsemi/Android-Scanner-Compat-Library #70: Question: how to detect adapter state change inside scanner BroadcastReceiver?

acutetech, 2019-10-09:
I have a sample app with one Activity and one BroadcastReceiver. The Activity starts scanning and BLE devices are detected in the BroadcastReceiver. If the user turns off the BLE adapter then scanning stops (not unreasonably). However when the BLE adapter is turned on again the scanning does not resume.

I have added a separate BroadcastReceiver inside the Activity which successfully detects the BLE adapter state changes, and I use this to restart scanning. However it seems more sensible to manage this within the  BroadcastReceiver, autonomously. 

How can I add BLE adapter state detection to the scanner BroadcastReceiver?

philips77, 2019-10-10:
You have to register the same broadcast receiver to the scanner state changes. If you have this receiver registered in Android manifest, I think this is not possible. You can't register to Bluetooth scanner changes in Android manifest, have to do this in runtime. But you may call the scanner receiver `onReceive` method from the other receiver and handle both events in one place.

philips77, 2019-10-10:
Regarding Automatic resuming, I will think about it. But I see a lot of problems. First, the original API doesn't do this. Second, what if a long time passed since Bluetooth was disabled. Or even if the phone was restarted. Resuming would work sometimes, but not always.

acutetech, 2019-10-10:
OK, thanks. This code inside my Activity works:
```
        BroadcastReceiver broadcastReceiver = new BleReceiver();    // My Receiver class, extends BroadcastReceiver
        IntentFilter belAdapterChange = new IntentFilter(BluetoothAdapter.ACTION_STATE_CHANGED);
        registerReceiver(broadcastReceiver, belAdapterChange);      // Now .ACTION_STATE_CHANGED events arrive on onReceive()
```
Then in my Receiver:

```
public void onReceive(Context context, Intent intent) {

        switch (intent.getAction()) {

            // Look at BLE adapter state
            case BluetoothAdapter.ACTION_STATE_CHANGED:
                final int state = intent.getIntExtra(BluetoothAdapter.EXTRA_STATE, BluetoothAdapter.ERROR);
                switch (state) {
                    case BluetoothAdapter.STATE_OFF:
                        Log.d(TAG, "BLE off");
                        break;
                    case BluetoothAdapter.STATE_TURNING_OFF:
                        Log.d(TAG, "BLE turning off");
                        break;
                    case BluetoothAdapter.STATE_ON:
                        Log.d(TAG, "BLE on");
                        break;
                    case BluetoothAdapter.STATE_TURNING_ON:
                        Log.d(TAG, "BLE turning on");
                        break;
                }
              // other events
}
```

philips77, 2021-06-04:
I'm closing it due to inactivity.

closed 2021-06-04 by philips77