nordicsemi/Android-Scanner-Compat-Library #72: Applying filters with Service Uuid sometimes not return results.

waqarul, 2019-10-23:
Hi,

I'm currently trying to set some ScanFilters while registering my BLEScanner to filter on service UUID.

I try in the following way:

    if (getIsUuidFilterEnabled()) {    
    
        val scanFilter = ScanFilter.Builder().setServiceUuid(ParcelUuid(TruvanBLEManager.UUID_SERVICE)).build()
            val filters = arrayListOf<ScanFilter>(scanFilter);
            scanner.startScan(filters, settings, truvanScanCallback)
        } else
            scanner.startScan(null, settings, truvanScanCallback)
    }

The problem occurs **when I frequently enable/disable the filter** due to which 

    onBatchScanResults(results: List<ScanResult>)
**I get the results size = >0**

Is there any solution for that?

philips77, 2021-06-04:
Hi,
Sorry for 1.5 year delay. Did you manage to solve the problem?

rokorec31, 2021-07-20:
Hi P

> Hi,
> Sorry for 1.5 year delay. Did you manage to solve the problem?

Hi Philips77,
I still have this problem on v1.5.0, and it occurs when frequently startScan/stopScan. Any solution now?

thanks for help!

philips77, 2021-07-21:
I am not aware of any. I hoped it automagically started to work.
Does it happen on all phones? On all Android versions? Did you try using the native scanner?

closed 2021-09-29 by philips77