nordicsemi/Android-Scanner-Compat-Library #73: ScanCallback.onScanFailed errorCode = 2

Ivan-Stashak-CardinalPeak, 2019-10-24:
All of sudden scanning started failing immediately with an errorCode of 2. This is happening on a Samsung S8 running Android 9. I've tried rebooting the phone and toggling bluetooth on and off.

Here's how I start a scan:

val scanner = BluetoothLeScannerCompat.getScanner()
val settings = ScanSettings.Builder()
                .setLegacy(false)
                .setScanMode(ScanSettings.SCAN_MODE_LOW_LATENCY)
                .setReportDelay(1000L)
                .setUseHardwareBatchingIfSupported(true)
                .build()

scanner.startScan(null, settings, scanCallback)

handler.postDelayed({
     stopScan()
}, 30000L)

Has anybody dealt with this?

philips77, 2019-10-24:
The error 2 seems to be this: https://developer.android.com/reference/android/bluetooth/le/ScanCallback.html#SCAN_FAILED_APPLICATION_REGISTRATION_FAILED
Do you have the `BLUETOOTH_ADMIN` permission? I think this happens when you try to start and stop scanning too often, but perhaps there's plenty of other reasons as well. You are saying that toggling Bluetooth didn't help?

Ivan-Stashak-CardinalPeak, 2019-10-24:
Ah, I need to reacquire ACCESS_FINE_LOCATION permissions, because I uninstalled my app... duh.

if (ContextCompat.checkSelfPermission(this.getApplicationContext(),
        android.Manifest.permission.ACCESS_FINE_LOCATION) == PackageManager.PERMISSION_GRANTED) {
} else {
    ActivityCompat.requestPermissions(this, new String[]{android.Manifest.permission.ACCESS_FINE_LOCATION},
            REQUEST_LOCATION_ENABLE_CODE);
}

closed 2019-10-24 by Ivan-Stashak-CardinalPeak

trix0, 2020-10-07:
Just to add to this You were right about this happening when you try to start and stop scanning too often.