nordicsemi/Android-Scanner-Compat-Library #74: unable to parse scan record when using startScan(filters, settings, callback)

waqarrashid33, 2019-10-24:
Hello everyone, 

I am having some issue with Android BLE and I tried using the standard Android BLE API and this one as well but the results are the same.

We have some devices that we can only see in the nRF Connect android app when we enable the `Offloaded scan batching`. If I am not wrong, this can be achieved by applying some delay using the setReportDelay() method in setting builder. Without any settings and filters I can see Bluetooth devices but when I try to apply any settings or filters, it doesn't show anything and I see the following error messages in logcat:
```
// some actual data is being replaced by x, y, z
unable to parse scan record: [x, y, z]
unable to parse scan record: [x, y, z]
unable to parse scan record: [x, y, z]
unable to parse scan record: [x, y, z]
```

Does anyone knows how to solve it or anyone have any idea how to achieve what nRF Connect app is doing when you enable `Offloaded scan batching`.

Thanks you very much. If anyone interested, I can provide you some example code that reproduce the problem. I would have shared it already but I don't know if its a bug or related to something obvious that I am doing wrong.

Cheers.
Waqar Rashid

philips77, 2019-10-24:
Hi,
For me this looks like an issue with the advertising data on your peripheral. Are you sure the structure of the advertising packet is correct? You may try with nRF Connect for PC if you have an nRF5 DK. From version.. 7? Android started validating advertising records and in case of a failure such packet is discarded. Perhaps, this validation is skipped when offloaded batching is used (you are correct, offloaded batching is enabled when you set `setReportDelay(...)` to something > 0). 

The correct structure of the advertising packet and scan response should have number of Length-Type-Value triples, where Length is a length of Type+Value, and Type has length 1. Android does not care about the types, but the lengths need to be correct. Single AIR can't be split between advertising data and scan response, as far as I know.

nRF Connect doesn't do magic with offloaded scan batching is enabled. It just sets `setReportDelay(500)`.

waqarrashid33, 2019-10-24:
I have an nRF5 DK. I will try to make it work and will ask here I will probably need some help with it. I am new to BLE and hadn't worked with any nordic device before.

philips77, 2019-10-25:
No, we use no filters and set report delay to 500. Remember, when you use this library, not to say `useHardwareBatchingWhenSupported(false)`. Then we use raw data from the ScanRecord.

waqarrashid33, 2019-10-25:
That means I shouldn't include anything about `useHardwareBatchingWhenSupported(false)`?
I tried using just the 

```
        settings = new ScanSettings.Builder()
                .setReportDelay(1000)
                .build();
```
But had no success. It still drops the adverts.

waqarrashid33, 2019-10-25:
For me, No useHardwareBatchingWhenSupported or true or false gives same result. No scan result and dropped advertisement packets.

waqarrashid33, 2019-10-25:
Is there any dependency with version of Android? What is the min version of android that supports offloaded scan batching?

philips77, 2019-10-25:
Offloaded batching is supported since Lollipop (Android 5), but not on all phones. Check `BLuetoothAdapter.isOffloadedBatchingSupported()`.

waqarrashid33, 2019-10-25:
It says yes but still would try to parse it automatically at the lower level and discard it. I got the firmware of the peripheral so hopefully I would be able to disable it there. We don't really need it, its just something the manufacturer did it.

waqarrashid33, 2019-10-25:
I sent you an email with the discarded advertisement data. Can you have a look at it if you can find a problem there?

philips77, 2021-06-04:
Hi,
Sorry for 1.5 year delay. I hope you managed to solve the issue. I'm closing it due to inactivity.

closed 2021-06-04 by philips77