nordicsemi/Android-Scanner-Compat-Library #75: No results found on scan

btoconnor, 2019-10-25:
I'm fairly confident I'm doing something wrong, but I'm having a hard time pinning it down.

I have a button that starts an LE scan.  I have log lines setup in my callback (for both `onScanResult` and `onBatchScanResults`.  `onBatchScanResults` log line shows up, but it consistently shows a `results.size` of 0.
I can open the reference NRF Toolbox app, and that application finds a number of BLE devices (as expected, there's several in the vicinity.

Some reference code here:
```kotlin
    private fun scanBLEDevices() {
        val scanner = BluetoothLeScannerCompat.getScanner()
        val settings = ScanSettings.Builder()
            .setLegacy(false)
            .setScanMode(ScanSettings.SCAN_MODE_LOW_LATENCY)
            .setReportDelay(1000)
            .setUseHardwareBatchingIfSupported(true)
            .build()

        val filters = mutableListOf<ScanFilter>()
        scanner.startScan(filters, settings, scanCallback)
    }
```

The callback:
```kotlin
    private val scanCallback = object : ScanCallback() {
        override fun onScanFailed(errorCode: Int) {
            Log.i("MYTAG", "error")
        }

        override fun onScanResult(
            callbackType: Int,
            result: ScanResult
        ) {
            Log.i("MYTAG","scan result!")
        }

        override fun onBatchScanResults(results: MutableList<ScanResult>) {
            Log.i("MYTAG", "Batch scan results: ${results.size}")
        }
    }
```
I get a fairly consistent stream of log messages like this: `Batch scan results: 0` until I stop the scan.

philips77, 2019-10-25:
Did you request for the location permission (fine or coarse) in runtime? Does your app declare them in the manifest, together with Bluetooth admin permission?

btoconnor, 2019-10-25:
Ah yes sorry, I forgot to mention that.

I have requested those permissions and they are granted for the application, both location and Bluetooth admin in the manifest. Location was granted during runtime

btoconnor, 2019-10-25:
To be explicit, here's what I've got in the manifest file:

```xml
    <uses-permission android:name="android.permission.BLUETOOTH" />
    <uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
    <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
```

I've got a button that requests `android.permission.ACCESS_FINE_LOCATION` on the activity, and that's been granted.  I then have another button that starts the scan, and that's where I get the log line showing 0 results.

Can provide more information if that's relevant

tompbsmith, 2019-10-26:
I have the same issue. I have the permissions BLUETOOTH, BLUETOOTH_ADMIN and ACCESS_FINE_LOCATION and a request for ACCESS_FINE_LOCATION is made in the Activity which I check is granted before I set up the Scanner. The ScanCallback onBatchScanResults() method is hit consistently with results being a list of size zero.

Is that basically the same as you @btoconnor ?

btoconnor, 2019-10-26:
Yeah that sounds exactly like what I'm seeing

philips77, 2019-10-27:
Do you have Location enabled on quick settings?

tompbsmith, 2019-10-27:
What do you mean by quick settings? I've got it enabled by finding my app in Settings>Apps if that's what you mean. Also it's checked at runtime as I said above.

Sorry for jumping on your thread bt, I can create a new one if that's less confusing.

btoconnor, 2019-10-27:
I have it on as well.

It sounds like we might be experiencing the same thing, so until we have evidence telling us otherwise sharing a thread here seems to make sense to me.

philips77, 2019-10-27:
Quick settings are those above notifications, that you drag from the top edge of the screen. There's Location there. It has to be enabled to be able to scan for Bluetooth LE devices.

btoconnor, 2019-10-27:
Yeah - like I said, it's on:
[image]

Also, I'm able to see BLE devices via the reference application and other applications.  I also implemented a basic test case activity to scan via the native implementation, and I can see results via that method.

btoconnor, 2019-10-27:
This may be helpful - I just set `setReportDelay(1000)` on the native implementation, and I'm seeing the same results.  If I remove that from the settings builder, I get device callbacks for each result.  If I set report delay to 1000ms, I get a result set of 0 in the native implementation.

Additionally, if I set `setUseHardwareBatchingIfSupported(false)`, I get batch results via this library.

So at this point it would seem that maybe the hardware batching on my phone is not working correctly?

philips77, 2019-10-27:
Did you try setting the delay to something longer, like 6s?

btoconnor, 2019-10-27:
I just switched to a delay of 6000ms.

When using hardware batching, I consistently get 0 results back.  If I set `.setUseHardwareBatchingIfSupported(false)`, I get results as expected.

philips77, 2019-10-28:
Yes, because with this flag set to false the library scans without a delay and just aggregates the results. Gives better and reliable results, but doesn't use offloaded batching, so it's battery inefficient.

ando0689, 2019-11-11:
I have exact same issue on Pixel device with Android 10, and the interesting thing is that it starts to find devices only when I set `setUseHardwareBatchingIfSupported(TRUE)` but with `FALSE` it does not find devices. 

Also I had to increase the scan time to 4s which is not good...
Do you know if this is a hardware specific issue or what? 

P.S. I noticed that nRF Toolbox app is also not finding any device with that phone, but nRF Connect finds..

philips77, 2019-11-11:
Remember, that on Android 10 you need to request Location Fine Permission, not Location Coarse. Otherwise you'll not get any results.

ando0689, 2019-11-13:
@philips77 Thanks! That solved my issue. Android 10 requires Fine Location permission for Bluetooth scanning.

RishikeshSreekumar, 2020-05-10:
I am also facing the same issue of getting 0 results and it is not working even when i increased the delay even to 10000ms

philips77, 2020-09-21:
Related #89 and #90

closed 2021-06-04 by philips77