nordicsemi/Android-Scanner-Compat-Library #76: How to scan for peripherals that have a given service UUID as a key in their service data?

sarbogast, 2019-11-11:
I'm trying to scan for peripherals that advertise a key-value pair in their service data where the key is a given service UUID and the value is a unique identifier, different for each peripheral. Those peripherals only use serviceData, they do not advertise the service UUID in serviceUuids.

Here is what I tried so far:
```kotlin
AsyncTask.execute {
       val scannedBleServices = mapOf(
            "ABC" to ParcelUuid(UUID.fromString("00001234-5678-9012-3456-789012345678"))
       )
       val scanFilters = scannedBleServices.values.map {
           ScanFilter.Builder().setServiceData(it, ByteArray(0)).build()
       }
       val scanSettings = ScanSettings.Builder()
                .setLegacy(false)
                .setScanMode(ScanSettings.SCAN_MODE_LOW_LATENCY)
                .setReportDelay(0L)
                .setCallbackType(ScanSettings.CALLBACK_TYPE_ALL_MATCHES)
                .setMatchMode(ScanSettings.MATCH_MODE_AGGRESSIVE)
                .setNumOfMatches(ScanSettings.MATCH_NUM_ONE_ADVERTISEMENT)
                .build()
        BluetoothLeScannerCompat.getScanner().startScan(
                scanFilters,
                scanSettings,
                leScanCallback
        )
}
```

But with that, the callback is never called. Note that if I pass `null` as a first parameter to `startScan`, my callback does get called, which seems to indicate that I have all the permissions working properly, and I have just a problem with my filters.

philips77, 2019-11-12:
Hi, 
First of all, you should use this method: https://developer.android.com/reference/android/bluetooth/le/ScanFilter.Builder.html#setServiceData(android.os.ParcelUuid,%20byte%5B%5D,%20byte%5B%5D):
```kotlin
val scanFilters = scannedBleServices.values.map {
     ScanFilter.Builder().setServiceData(it, ByteArray(X), ByteArray(X)).build()
}
```
Then it will look for a Service Data of size X-bytes, but none of the bits will matter, as the mask is set to X-bytes of 0. I think this should work. X should be the length of your unique identifier.

The other issue with this is that iOS only allows to scan in background for devices advertising Service UUID. If you want to use your device on iOS as well, and you want to allow it to be scanned when the app is in background, then you need to add Service UUID with the same UUID.

sarbogast, 2019-11-12:
As a matter of fact, on iOS, I'm able to set a filter on serviceUUID and my peripherals show up in the background even if I'm only advertising this service UUID in serviceData apparently. I haven't found a way to filter on serviceData in CBCentralManager anyway.

About Android, I tried your suggestion like so:
```
val scanFilters = Settings.scannedBleServices.values.map {
        ScanFilter.Builder().setServiceData(it, ByteArray(4), ByteArray(4)).build()
}
```

Since my unique ID is 4 bytes long (32 bits), but I'm still not getting any callback. Here is what my device looks like in nrfConnect:
[image]

Of course I changed the service UUID in the question above but the one I'm filtering for matches the one in the screenshot.

philips77, 2019-11-14:
> As a matter of fact, on iOS, I'm able to set a filter on serviceUUID and my peripherals show up in the background even if I'm only advertising this service UUID in serviceData apparently.

Oh, good to know. Did you try setting `setUseHardwareFilteringIfSupported(false)` and see the difference? Is it an issue in the library or just the system does not filter well?
https://github.com/NordicSemiconductor/Android-Scanner-Compat-Library/blob/8df7799441def4db80b7c9055ff99ecb57d9dad3/scanner/src/main/java/no/nordicsemi/android/support/v18/scanner/ScanSettings.java#L511

sarbogast, 2019-11-14:
That seems to have done the trick, thanks a lot.

closed 2019-11-14 by sarbogast

philips77, 2019-11-14:
But I wouldn't recommend this. It just moves filtering to the library, but the offloaded filtering isn't used, so this has huge impact on the battery. I'd recommend finding a way to make it work without this method. You may also check on some other phones, perhaps it's just the one that has filtering issues.

sarbogast, 2019-11-14:
The phone I'm testing with is pretty standard and modern: Samsung Galaxy S9. So if that's what it takes for it to work, I don't see a lot of other choices.