nordicsemi/Android-Scanner-Compat-Library #77: Nordic thingy:52 unable to get RSSI values

Xiryl, 2020-02-19:
Hi, everybody,

I managed, through the Nordic library for Android `thingylib` ([github](https://github.com/NordicSemiconductor/Android-Nordic-Thingy/tree/master/thingylib)) , to connect to thingy52 and receive in the callback listener `ThingyListener`  the information about the device (e.g. `onGravityVectorChangedEvent`, `onAccelerometerValueChangedEvent`, `onGyroscopeValueChangedEvent`, and so on...).

What I can't do (and I can't find how to do) is how I can get the information about the connection with the device, exactly the RSSI values. 

Using the [BluetoothLeScannerCompat](https://github.com/NordicSemiconductor/Android-Scanner-Compat-Library), inside the `onBatchScanResults` scan callback, I detect the thingy52 bluetooth device, and their RSSI value, but if I set a high scan frequency, for example `.setReportDelay(10)`, it almost always does not detect devices.

```java
 ScanSettings scanSettings = new ScanSettings.Builder()
                .setScanMode(ScanSettings.SCAN_MODE_LOW_LATENCY)
                .setReportDelay(10) // frequency of scan
                .setUseHardwareBatchingIfSupported(false)
                .build();
```

My need is to get at a frequency of `10Hz`, or `100Hz` the RSSI value of the nordic thingy52, which I can't get from the `ThingyListener` events. 

How can I do it? 
Thank you in advance.

philips77, 2020-02-20:
For accurate RSSI values you can't use `setReportDelay`. Set it to 0 or remove from the code. First of all, batching works quite ok for delays > 5 seconds. Secondly, the RSSI you get there is an average/last/first/no one knows what. Without report delays you'll get a `onScanResult` callback instead. One for each advertisement received.

Xiryl, 2020-02-21:
Thanks @philips77. Are you aware of a way to get the RSSI value once connected to the nordic?
For example, in one of the callbacks provided by the listener, there is a way to capture the RSSI value? 

```java
 @Override
    public void onGyroscopeValueChangedEvent(BluetoothDevice bluetoothDevice, float x, float y, float z) {
      // read rssi val here ?
      Log.d(TAG, "onGyroscopeValueChangedEvent: " + bluetoothDevice.getName());
    }
```

philips77, 2020-02-24:
Hello,
Yes, you need to call [BluetoothGatt#readRemoteRssi()](https://developer.android.com/reference/android/bluetooth/BluetoothGatt#readRemoteRssi()). The RSSI will be returned asynchronously using [BluetoothGattCallback#onReadRemoteRssi(BluetoothGatt, int, int)](https://developer.android.com/reference/android/bluetooth/BluetoothGattCallback#onReadRemoteRssi(android.bluetooth.BluetoothGatt,%20int,%20int)).

Xiryl, 2020-02-24:
Thanks @philips77 ! Seems working.
With the default `thingySdkManager ` settings I seem to have the nordic listener callback every 200ms or so. Is there a method to tell `thingySdkManager` to get the data with some frequency?

philips77, 2020-02-25:
Please, open an issue in the thingy project for this. Let's keep things tidy. I'm closing this issue.

closed 2020-02-25 by philips77