nordicsemi/Android-Scanner-Compat-Library #78: IllegalStateException on stop scan

jtfiskar, 2020-03-09:
This is my MainActivity onStop method which sometimes shows me crash reports (java.lang.IllegalStateException:) indicating that scanCallback would be null. I don't understand how it is possible to be null. Would it be safe to try catch here since I dont have any other solution?

public void onStop() {
        Log.w(TAG, "onStop");

        if(mScanning) {
            Log.i(TAG, "stop scanning");
            stopScan();
        }
        if(mConnected) {
            Log.i(TAG, "disconnect");
            mBleManager.disconnect();
        }

        super.onStop();
}

philips77, 2020-03-09:
Do you experiencing it when the Bluetooth is disabled?
I think it's safe to wrap in try-catch, as it's cleaning-up, after all.

jtfiskar, 2020-03-09:
Yes you are correct, if bluetooth is disabled while scanning then this happens. Thank you.

closed 2020-03-09 by jtfiskar