nordicsemi/Android-Scanner-Compat-Library #79: No callback response, no result. Samsung Galaxy A10/A50

snakecatcher, 2020-03-12:
Hi there.
I need help.
Here is my code:

        BluetoothLeScannerCompat scanner;
        scanner = BluetoothLeScannerCompat.getScanner();
        ScanSettings scanSettings = new ScanSettings.Builder().
                setLegacy(true).
                setScanMode(ScanSettings.SCAN_MODE_LOW_LATENCY).
                setReportDelay(0).
                build();

        List<ScanFilter> scanFilters = new ArrayList<>();
        scanFilters.add(new ScanFilter.Builder().setServiceData( new ParcelUuid(UUID.fromString("0000180f-0000-1000-8000-00805f9b34fb")), new byte[0]).build());
        scanFilters.add(new ScanFilter.Builder().setServiceData( new ParcelUuid(UUID.fromString("00000002-0000-1000-8000-00805f9b34fb")), new byte[0]).build());
        scanFilters.add(new ScanFilter.Builder().setServiceData( new ParcelUuid(UUID.fromString("00000001-0000-1000-8000-00805f9b34fb")), new byte[0]).build());
        scanFilters.add(new ScanFilter.Builder().setServiceData( new ParcelUuid(UUID.fromString("00000008-0000-1000-8000-00805f9b34fb")), new byte[0]).build());

       scanner.startScan(scanFilters, scanSettings, scanCallback);

It's my callback:
 
    private ScanCallback scanCallback = new ScanCallback() {
        @Override
        public void onScanResult(int callbackType, ScanResult result) {
            super.onScanResult(callbackType, result);
            String deviceNumber = "";
            if(result.getDevice()!=null && !TextUtils.isEmpty(result.getDevice().getName()) && result.getDevice().getName().startsWith("01")) {
                deviceNumber = result.getDevice().getName();
                Log.i("LogTag", deviceNumber);
            }
        }

        @Override
        public void onScanFailed(int errorCode) {
            super.onScanFailed(errorCode);
        }
    };

Manifest:

    <uses-permission android:name="android.permission.BLUETOOTH" />
    <uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
    <uses-feature android:name="android.hardware.bluetooth_le" android:required="true"/>
    <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
    <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />

Gradle

    implementation 'no.nordicsemi.android.support.v18:scanner:1.4.3'

On the Samsung GalaxyA10 (Android 9) - callback not called.

snakecatcher, 2020-03-12:
If I remove filters like in #37, the code works on all phones . 

 scanner.startScan(null, scanSettings, scanCallback);

But with filters, in my case - faster. Tell me , is it possible to work with filters?

philips77, 2020-03-12:
Hello,
First of all, you need to check whether your phone supports offloaded filtering using [isOffloadedFilteringSupported()](https://developer.android.com/reference/android/bluetooth/BluetoothAdapter#isOffloadedFilteringSupported()). But I guess it does, and even if not, the library would switch to emulating the same behaviour. 
Do you get callbacks on other phones than A10?

snakecatcher, 2020-03-12:
Yes, I get callbacks to other phones.

```
BluetoothManager bluetoothManager = (BluetoothManager)getApplicationContext().getSystemService(BLUETOOTH_SERVICE);
BluetoothAdapter bluetoothAdapter = bluetoothManager.getAdapter();
if(bluetoothAdapter.isOffloadedFilteringSupported()){
  Toast.makeText(MainActivity.this, "Supported", Toast.LENGTH_LONG).show();
  scanner.startScan(scanFilters, scanSettings, scanCallback);
} else {
  Toast.makeText(MainActivity.this, "Not Supported", Toast.LENGTH_LONG).show();
  scanner.startScan(null, scanSettings, scanCallback);
}
```

On A10 - bluetoothAdapter.isOffloadedFilteringSupported() - return true.

philips77, 2020-03-12:
Well... Then it looks like an issue on the controller firmware on this phone. That's why we added `useHardwareFilteringIfSupported(...)`. You may enforce cpu filtering on such phones.

philips77, 2020-03-13:
Also, try some other filtering data. Check if a single filter works, filtering by device address or name, etc. If you could determine where is the edge of what's supported.

snakecatcher, 2020-03-13:
I tried "setUseHardwareFilteringIfSupported(true)", but it didn't help.
If "scanFilters.size = 0" - it's work.
If "scanFilters.add(new ScanFilter.Builder().setDeviceName("01000009").build());" - it's work.
Single filter with UUID - didn't work.

philips77, 2020-03-15:
On the phone's that you can't make it work use `setUseHardwareFilteringIfSupported(false)` so the library will emulate the correct behavior. But this will have impact on battery, so make sure you set this parameter to true know on the phones you know they don't work.

snakecatcher, 2020-03-18:
It's work!!! :)
Thank you so much for your help!!!
```
 ScanSettings scanSettings = new ScanSettings.Builder().
                setUseHardwareFilteringIfSupported(false).
                setScanMode(ScanSettings.SCAN_MODE_LOW_LATENCY).
                setReportDelay(0).
                build();
```

closed 2020-03-18 by snakecatcher