nordicsemi/Android-Scanner-Compat-Library #81: Scanner returns results with inconsistent delays

mihibo5, 2020-03-26:
I have the following code:
```
BluetoothLeScannerCompat scanner = BluetoothLeScannerCompat.getScanner();
ScanSettings settings = new ScanSettings.Builder()
    .setLegacy(false)
    .setScanMode(ScanSettings.SCAN_MODE_LOW_LATENCY)
    .setReportDelay(0)
    .setUseHardwareBatchingIfSupported(true)
    .build();
scanner.startScan(null, settings, scanCallback);
```

The scanner works, but it behaves differently on different phones. Samsung XCover 4 triggers callback 4 times per second, but Huawei phones trigger callbacks with inconsistent frequencies. At first the frequency is once per second, over short period of time (30 seconds at most) the delay becomes random between 1 second to 10 seconds. What do I have to do?

philips77, 2020-03-27:
Hello,
When report delay is set to 0, the results are returned immediately after an advertising packet is scanned (and optional scan response is received). There's no fixed interval. It depends on number of advertising BLE devices in range and their advertising interval.
When the report delay is set to > 0, it all depends on the value you call `setUseHardwareBatchingIfSupported` with. If offloaded batching is used (this method called with parameter *true*), the report delay should be 5000 or more. We found, that with shorter delay the results don't come in regular intervals and the first one comes delayed a lot. With delay 5 seconds or more, the results are quite reliable, but as packets from one device may be merged into a single result, there may be some differences comparing to what would you receive with delay 0. For example, all results may have the same timestamp of the time they were returned, not scanned. Also, we've heard that some devices don't show up (it depends on the adv data). Devices advertising with PHY Coded may not be received, or may have set their primary PHY set to LE 1M. I would recommend testing on multiple phones that the results from your device are returned correctly.
When this parameter is set to *false*, than the library will scan with report delay 0, which drains battery more quickly, will aggregate the results and return them in regular interval. Results will have the same values as scanned without report delay. This mode, however, should not be used for long-lasting background scanning, as the phone battery will quickly die. It may be improved by setting scan mode to `SCAN_MODE_LOW_POWER`, which scans for 500 ms every 5 seconds.

closed 2020-03-27 by philips77

mihibo5, 2020-03-30:
I have set the beacon in nRF Connect to transmit every 250 ms, but I recieve it randomly. I have also discovered the difference between Samsung and Huawei phones is that Samsung has Bluetooth 5.0 and Huawei phones have Bluetooth 4.2. Could that be one of the resasons?

philips77, 2020-03-30:
When advertising interval is set to X ms, that means that the device will wait this time and then start advertising on 3 channels, one after another. It should also add some random delay each time to lower a risk of packet collision. That means, that your phone can pick any of the 3 packets, with any random delay (there's a range of possible delays, don't remember how long. Up to few ms, or less). So if your advertising interval is set to 250 ms, you should receive packets each 250-260 ms, or 500-510 (if all packets in a single advertisement were missed), etc.
That's why nRF Connect doesn't show exact value, but approximate. It needs to calculate the true advertising interval based on few packets received.

All Bluetooth chipsets should be backwards compatible, but some lower end devices have cheaper components, so there may also be issues. People say: "It works on my iPhone, but doesn't on my Samsung A5, why?!" But iPhones aren't low end devices, so they can't be compared with such.