nordicsemi/Android-Scanner-Compat-Library #85: Unable to scan for other Android phones

Simonida, 2020-05-06:
I've successfully used this library for scanning some hardware devices (BT chips ESP32, Sierra Wireless, etc.).
But, I'm unable to detect other Android phones (that have Bluetooth and Location enabled).

Is there any specific configuration necessary to be added, or this is not possible to be achieved?

This is my implementation:

```
public synchronized void startScan() {
		final ScanSettings settings = new ScanSettings.Builder().setScanMode(ScanSettings.SCAN_MODE_LOW_LATENCY)
			.setReportDelay(1000)
			.setUseHardwareBatchingIfSupported(false)
			.build();
		
		BluetoothLeScannerCompat scanner = BluetoothLeScannerCompat.getScanner();
		try {
			scanner.startScan(null, settings, scanCallback);
			Log.d(LOG_TAG, "BLE start scan.");
		} catch (IllegalArgumentException e) {
			Log.d(LOG_TAG, e.getMessage());
		}
	}
```

Thanks!

philips77, 2020-05-06:
How do you advertise on the phone?

Simonida, 2020-05-06:
I don't do any advertising on the other phones.
The idea was to detect the all phone devices that have just Bluetooth and Location enabled.

I was investigating today and it looks like phones needs to do BLE advertising (through some specific app) in order to be discoverable via BLE scanning, is that right?

Sorry for opening the issue here, but the official documentation is somewhat confusing regarding peripheral mode...

philips77, 2020-05-06:
Yes, you need to advertise in an app. We recommend nRF Connect for Android, available on Google Play, where you can customize the advertising packet in UI.

philips77, 2020-05-06:
On the other hand, iOS devices do advertise without any app, as they use it for AirDrop. Apps can customize and add their data to the packet, but even without them the phone will advertise. Same with Macs with support for BLE.

closed 2020-05-06 by philips77

Simonida, 2020-05-06:
@philips77 Thank you very much! You've helped me a lot with your answers.

trycatchx, 2022-05-07:
@philips77 why  My  iphone can find my other android phone bluetooth, but this demo cannot find it ?

`
		// Scanning settings
		final ScanSettings settings = new ScanSettings.Builder()
				.setScanMode(ScanSettings.SCAN_MODE_LOW_LATENCY)
				.setReportDelay(500)
				.build();

		final BluetoothLeScannerCompat scanner = BluetoothLeScannerCompat.getScanner();
		scanner.startScan(null, settings, scanCallback);
		scannerStateLiveData.scanningStarted();`