nordicsemi/Android-Scanner-Compat-Library #89: ScanCallback does not call on Android 10

Stalker11, 2020-09-21:
Hello everyone.
I try to scan BLE devices on Android 10, but ScanCallback does not call. All permissions are granted. 
On previous versions Android it works fine.
Used library version no.nordicsemi.android.support.v18:scanner:1.1.0 and no.nordicsemi.android:ble:1.2.0
After migration to no.nordicsemi.android.support.v18:scanner:1.4.3 and no.nordicsemi.android:ble:2.2.4 I still have this problem.
Could you help me, how I can solve it?

philips77, 2020-09-21:
You need to enable Location Service as well. You may find it in the quick settings under name Location. When disabled, no app have a access to any location-related features, which includes scanning for BLE devices as well.

Stalker11, 2020-09-21:
Location Service is enabled and all permissions granted. But ScanCallback does not find any devices only on Android 10.

philips77, 2020-09-21:
Then I have no idea. Try restarting Bluetooth, the phone or checking another phone with Android 10.

philips77, 2020-09-21:
Do you request `ACCESS_FINE_LOCATION`, not `ACCESS_COARSE_LOACTION`?

Stalker11, 2020-09-21:
I have tested on different devices (Huawei, Samsung, Xiaomi). None of them does works with BLE. 
Permissions:

<uses-permission android:name="android.permission.ACCESS_BACKGROUND_LOCATION"/>
<uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
    <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
    <uses-permission android:name="android.permission.BLUETOOTH" />
    <uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />

philips77, 2020-09-21:
Do you request `android.permission.ACCESS_FINE_LOCATION` in [runtime](https://developer.android.com/training/permissions/requesting)?

Stalker11, 2020-09-21:
yes

Stalker11, 2020-09-22:
I have no idea what was wrong, but I have changed requiring runtime permissions in the migration app and it works:
`private val locationPermission = if (Build.VERSION.SDK_INT >= 29)
        permission.ACCESS_FINE_LOCATION else permission.ACCESS_COARSE_LOCATION`

closed 2020-09-22 by Stalker11

wykeenjenga, 2023-01-18:
Hey @Stalker11 did you find any solution?