nordicsemi/Android-Scanner-Compat-Library #93: Scanner

malashkevich, 2021-02-03:
I'm using this code to scan for nordic BLE chip

        isScanning = scanning;
        scanResultSet.clear();

        ScanSettings settings = new ScanSettings.Builder()
                .setLegacy(false)
                .setScanMode(ScanSettings.SCAN_MODE_LOW_LATENCY)
                .setReportDelay(1000)
                .setUseHardwareBatchingIfSupported(true)
                .build();

        scanCallback.setDeviceConsumer(deviceConsumer);

        scanner.startScan(null, settings, scanCallback);

It works pretty well except for one case.
When an Android device never met the particular BLE chip and I'm trying to scan using the code above, the app is not able to find that device.
If I switch to the NRF Connect app or will try to scan via OS settings the device will appear.
When I return back to my app it will also appear.

This happens really for the first time for the device. Maybe you have some ideas about what it might be.

malashkevich, 2021-02-04:
Btw, is it possible to have access to one piece of NRF Connect source code (just regarding scanning)?
It would help me a lot.

philips77, 2021-02-04:
Hi,
Try setting report delay to 0, or use 
https://github.com/NordicSemiconductor/Android-Scanner-Compat-Library/blob/8f4907ed1b49439724e4c96a697ef0919289e16b/scanner/src/main/java/no/nordicsemi/android/support/v18/scanner/ScanSettings.java#L529
With value *false*. That could help, i think. Hardware batching works better with delay > 5 sec.

malashkevich, 2021-02-08:
@philips77 interesting but it seems that issue is resolved.
Do you have any idea why this is happening?

closed 2021-02-08 by malashkevich

philips77, 2021-02-08:
The offloaded batching is a low power mode on the BLE controller. It uses much less power, I suppose, than continuous scanning, but the low power comes with some cost. Also, I believe that the behavior may differ among different chip vendors.
Basically, if you want to scan for a short period of time, and user is waiting for the result, you should use no batching and low latency. If you plan to scan in background, of for long time, batching is what you may want, but then, as I said, use delays > 5 sec so the chip could gather more packets before reporting to the app.

This library can emulate batching when the above-mentioned flag is set to false. It scans with no batching (using more power, but you get more devices, as if you'd scan without batching), and combines them into batches, so your app doesn't need to do this.

malashkevich, 2021-02-08:
Sounds reasonable. Thanks for explanation!