nordicsemi/Android-Scanner-Compat-Library #94: NullPointerException while starting scanner on Android 12 DP3

Leo38, 2021-04-28:
Hello,

While testing our application on **Android 12 DP3**, we have observed a NullPointerException at scanner start. 

Investigating a bit, it comes from Google API changes on `ScanFilter.Builder.setDeviceAddress`. With last changes, calling it with null device address will always throw a **NullPointerException**

I have created an issue on Google issue tracker : [https://issuetracker.google.com/issues/186574846](https://issuetracker.google.com/issues/186574846)

If this issue is not fixed on Google side, it might be important to check for nullability before creating the ScanFilter :  [ScanFilter creation](https://github.com/NordicSemiconductor/Android-Scanner-Compat-Library/blob/8f4907ed1b49439724e4c96a697ef0919289e16b/scanner/src/main/java/no/nordicsemi/android/support/v18/scanner/BluetoothLeScannerImplLollipop.java#L197)

philips77, 2021-04-28:
Thank you very much! Awesome work!

mtrewartha, 2021-05-12:
Looks like someone at Google marked the issue as fixed today. Sounds like it'll be available in the next DP/Beta build.

arstagaev, 2021-05-13:
I have the same issue, on Pixel 4
Log:
```
Fatal Exception: java.lang.NullPointerException
       at com.android.internal.util.Preconditions.checkNotNull(Preconditions.java:147)
       at android.bluetooth.le.ScanFilter$Builder.setDeviceAddressInternal(ScanFilter.java:678)
       at android.bluetooth.le.ScanFilter$Builder.setDeviceAddress(ScanFilter.java:615)
       at android.bluetooth.le.ScanFilter$Builder.setDeviceAddress(ScanFilter.java:590)
       at no.nordicsemi.android.support.v18.scanner.BluetoothLeScannerImplLollipop.toNativeScanFilter(BluetoothLeScannerImplLollipop.java:210)
       at no.nordicsemi.android.support.v18.scanner.BluetoothLeScannerImplLollipop.toNativeScanFilters(BluetoothLeScannerImplLollipop.java:203)
       at no.nordicsemi.android.support.v18.scanner.BluetoothLeScannerImplLollipop.startScanInternal(BluetoothLeScannerImplLollipop.java:81)
       at no.nordicsemi.android.support.v18.scanner.BluetoothLeScannerCompat.startScan(BluetoothLeScannerCompat.java:158)
       at com.arstagaev.bluetoothapp15itelma.nordic.scan.NordicBleScanViewModel.startScan(NordicBleScanViewModel.kt:38)
       at com.arstagaev.bluetoothapp15itelma.nordic.scan.NordicBleScanActivity.onStart(NordicBleScanActivity.kt:105)
       at android.app.Instrumentation.callActivityOnStart(Instrumentation.java:1455)
       at android.app.Activity.performStart(Activity.java:7984)
       at android.app.ActivityThread.handleStartActivity(ActivityThread.java:3577)
       at android.app.servertransaction.TransactionExecutor.performLifecycleSequence(TransactionExecutor.java:221)
       at android.app.servertransaction.TransactionExecutor.cycleToPath(TransactionExecutor.java:201)
       at android.app.servertransaction.TransactionExecutor.executeLifecycleState(TransactionExecutor.java:173)
       at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:97)
       at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2151)
       at android.os.Handler.dispatchMessage(Handler.java:106)
       at android.os.Looper.loopOnce(Looper.java:201)
       at android.os.Looper.loop(Looper.java:288)
       at android.app.ActivityThread.main(ActivityThread.java:7695)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:981)
```

mtrewartha, 2021-05-19:
Bummer, looks like Android 12 Beta 1 didn't fix the issue.

philips77, 2021-06-04:
Fixed in 1.4.4.

closed 2021-06-04 by philips77