nordicsemi/Android-Scanner-Compat-Library #101: Best Practices for Long-Running Scanning

robbyice, 2021-06-19:
Hey! I'm working on an application for a client right now that uses this library to continuously scan for nearby bluetooth beacons. The component of the app that manages the scanning is a singleton that is scoped to the life of the Application - we use Hilt to manage this. When the user logs into the application, a foreground service is started that runs indefinitely until the user logs out of the application. Effectively this means our application process and this scanning component singleton will run for extremely long periods of time - multiple hours and possibly multiple days if users don't power off their devices at the end of their work day. For the purposes of this application we need scanning data always available at potentially any point in time - hence the nonstop scanning.

There has been an issue reported in the field wherein the application will suddenly stop receiving results from the scanning operation. This issue reportedly persists through both power cycling the device and/or reinstalling the application. After some number of hours, users would reinstall the application once more and the app would work normally. This issue does not seem to happen often or consistently in any manner - every once in awhile things just break for a period of time and then eventually start working again after users reinstall the app. This issue has only happened a few times - otherwise the app works just fine. Both of the users that have reported this issue are using a Pixel 3a device running Android 11. 

Through my research in investigating this issue I've read discussion around how, as of Android 7+, the OS will downgrade the scanning mode for a scan that runs longer than 30 minutes into an opportunistic scan. As I understand it, this means the application that initiated the long-running scan will no longer receive scan results unless there is another application with a non-opportunistic scan that detects devices that match the opportunistic scan's targeted filters. Initially I thought this was a promising discovery, but the app has consistently worked for multiple hours across multiple days without any issue - and when users did report this rare issue they had already been using the app successfully for several hours. There are no other applications users have running that would be scanning for these devices.

So my first question is this - is there something going on in this library that prevents scans longer than 30 minutes from being automatically downgraded? I've poked around a bit and haven't seen anything that seems to address this, compared to, for example, AltBeacon's Android-Beacon library which [seems to have code that explicitly accommodates](https://github.com/AltBeacon/android-beacon-library/blob/c659c772b11df2af015fc527a993c54cd6c95b0a/lib/src/main/java/org/altbeacon/beacon/service/scanner/CycledLeScanner.java#L584) this by stopping and restarting a scan before it hits that timeout. 

Assuming this library isn't doing anything to address this and it's expected that you manage your own stopping/restarting of a long running scan within the timeout period, I'm not sure why the current application would  _ever_ work when scanning for more than 30 minutes - it seems to be able to do this quite consistently. I've been unable to recreate this issue locally on my test device, despite leaving the scanning running for over 24 hours. The vast majority of the time, people in the field are able to use the app for multiple hours without issue. Is the OS downgrading the scanning not guaranteed to happen after the timeout but instead driven by resource consumption or some other factor? The research I've done suggests that it unconditionally downgrades the scan after 30 minutes.

In general I am wondering if there is any advice to be had when it comes to using this library for extremely long running scans. Android BLE seems to be a...uh...challenging ecosystem to work in so any help understanding best practices for using this library in this capacity or suggestions for debugging this issue would be immensely appreciated!

Similarly, if there's somewhere else I should post or look for help further understanding this issue, please let me know.

philips77, 2021-06-21:
Hello @robbyice,

First of all, thank you for sharing your finding.

This library is not doing anything to "workaround" this 30 minutes limit. The main goal here is to bring/emulate new features on old OS versions, not to bypass limitations. However, I understand that it would be a nice to have feature, so we will consider.

As to why your app may stop getting advertisements, I can see 2 possible reasons. First, like you said, is related to this downgrading of a scan to opportunistic. But that should be easily verifiable, e.g. by asking a customer to scan using another app. I, however, doubt it's this. More probably it may be related to releasing resources. In this library we used to have an issue #58 which caused the phone to stop reporting after 32 scans. It has been fixed now in 1.4.4+, but perhaps there is similar one related to other type of scanning, or in the Android system itself. Uninstalling the app could be the way of releasing the handles and making the app start scanning again.

Out of curiosity, what is the lowest API supported by your app and how do you scan in the background? Are you using 
https://github.com/NordicSemiconductor/Android-Scanner-Compat-Library/blob/cd137c118173ce15bfd53687be965b95bcc91db1/scanner/src/main/java/no/nordicsemi/android/support/v18/scanner/BluetoothLeScannerCompat.java#L268-L272? Did you do any test on how does emulating this way of scan impacts a battery on pre-Oreo versions? Did you compare the battery impact with your app running and without? If you could share, of course :)

robbyice, 2021-06-22:
Hi @philips77 , thanks for responding!

I was pretty sure there wasn't anything this library was doing that would affect the 30 minute timeout, so thanks for verifying that. I think it's reasonable to not expect this library to do anything special for long-running continuous scanning; it is as you said, intended to be used for emulating newer features on old OS versions. It might be worth mentioning something regarding a recommended approach for continuous scanning in the README, but that's up to you.

The lowest API version supported by the application is 21. I have yet to do any testing on the devices I have using an earlier Android version. All my attempts to recreate the issue have been on a Pixel 2 running Android 11. This is primarily due to the fact that the two users that reported this issue were using Pixel 3a also running Android 11. It is possible that other users experienced this issue on other devices/platform versions but nobody has reported anything yet. Unfortunately I don't have visibility into the distribution/count of devices and API versions people are using - I am hoping to try and get this information soon.

For starting the scanning, we aren't using the 'startScan' method with a PendingIntent - we're using the method that just takes a callback along with the settings and filters. This is roughly what our scanning component looks like:

```
@Singleton
class BluetoothScanner @Inject constructor(@ApplicationContext context: Context) {
    private val mScanner = BluetoothLeScannerCompat.getScanner()

    val devices = BroadcastChannel<ArrayMap<String, ScanResult>>(1)
    val device = BroadcastChannel<ScanResult>(1)

    private val defaultSettings = ScanSettings.Builder()
        .setScanMode(ScanSettings.SCAN_MODE_LOW_LATENCY)
        .setUseHardwareBatchingIfSupported(false)
        .build()

    private val defaultFilters = getFilters()

    var scanning: Boolean = false
    private val mDevices: ArrayMap<String, ScanResult> = ArrayMap()

    private val scanCallback = object : ScanCallback() {
        override fun onScanResult(callbackType: Int, result: ScanResult) {
            Timber.d("Discovered device ${result.device.address}")

            mDevices[result.device.address] = result
            GlobalScope.launch(Dispatchers.IO) { device.send(result) }
        }

        override fun onBatchScanResults(results: List<ScanResult>) {
            Timber.d("Discovered ${results.size} device(s)")

            if (results.isEmpty())
                return

            mDevices.clear()
            results.forEach {
                mDevices[it.device.address] = it
            }
            GlobalScope.launch(Dispatchers.IO) { devices.send(mDevices) }
        }

        override fun onScanFailed(errorCode: Int) {
            Timber.e("Bluetooth scan failed, err ${errorCode}")
            scanning = false
        }
    }

    fun start() {
        Timber.d("start")

        if (!isBluetoothEnabled) {
            Timber.d("bluetooth is not enabled, unable to start scanning...")
            return
        }

        if (scanning) {
            Timber.d("already scanning, skipping...")
            return
        }

        mScanner.startScan(defaultFilters, defaultSettings, scanCallback)
        scanning = true
    }

    fun stop() {
        Timber.d("stop")
        mScanner.stopScan(scanCallback)
        scanning = false
        mDevices.clear()
    }
}
```

Some issue around resource consumption was something I had initially considered when investigating this issue. However to clarify a bit, users have reported that reinstalling the app does **not** solve this problem. Users have reported power cycling the device, doing a reinstall of the application, and then continuing to experience this issue _immediately_ upon restarting the application. Once they reach this point, users will reinstall an older version of the application and use that version for awhile without issue. After some period of time, say a few hours, they will once again re-install the newer version of the app and then it will resume working normally.

A few potentially relevant bits of info on the older version of the app:
- The scanning code between the newer and older apps is identical, the new app is so far just a overhaul/redesign of the UI components. We haven't changed anything BLE related.
- The older app is signed with a different key and side-loaded onto the phone. The newer app is distributed on an internal testing track via the Google Play store.

A couple of my own thoughts:
- The fact that the issue persists across power cycling the device and reinstalling the app is particularly confusing, and makes me think that the issue has to be related to something the OS is doing
- The fact that using the old app for reportedly a few hours then results in things eventually returning to normal in the new app _kind of_ reinforces the whole OS-downgrading-the-scan-mode theory, but it doesn't really explain why the issue happens after 5-6 hours of use rather than 30 minutes. It also doesn't really explain why the issue is experienced so infrequently, with only 3 reported instances across 2 users over the last couple months.
- It's unclear if users uninstalled the old version of the application before returning to the new one to find it once again working.I'll have to ask people some more questions about this part of their experience.

robbyice, 2021-07-22:
Going to close this issue with a few notes on how my team has planned to move forward with this issue.

Firstly we are changing our scanning code to happen more intentionally - previously scanning would begin via a foreground service as soon as the application launched and would not stop until a user either logged out or the device was powered down. The new plan is to begin scanning operations specifically when we need the data, and adjust anything as needed to handle the fact that results will take some time to stabilize before they should be reported on.

Secondly, when we do scan, we'll be scanning in intervals rather than leaving the operation running continuously. Hopefully this will help prevent any issue with scanning results no longer being delivered in the event that a scanning operation needs to be run for a prolonged period of time.

Given how ephemeral and hard to recreate this issue is it's hard to say if these things will actually solve the problem, but hopefully if a few months pass without further incident reports we can be confident that this worked.

closed 2021-07-22 by robbyice