nordicsemi/Android-Scanner-Compat-Library #102: How to fix error: "Could not resolve no.nordicsemi.android.support.v18:scanner:1.5.0" ?

NanaDada, 2021-06-23:
The following errors started to show up when syncing the gradle files after upgrading from version 1.4.3 to 1.5.0:

```
A problem occurred configuring root project 'android'.
> Could not resolve all artifacts for configuration ':classpath'.
   > Could not resolve no.nordicsemi.android.support.v18:scanner:1.5.0.
     Required by:
         project :
      > No matching variant of no.nordicsemi.android.support.v18:scanner:1.5.0 was found. The consumer was configured to find a runtime of a library compatible with Java 11, packaged as a jar, and its dependencies declared externally but:
          - Variant 'releaseApiPublication' capability no.nordicsemi.android.support.v18:scanner:1.5.0 declares a library, and its dependencies declared externally:
              - Incompatible because this component declares an API of a component, with the library elements 'aar' and the consumer needed a runtime of a component, packaged as a jar
              - Other compatible attribute:
                  - Doesn't say anything about its target Java version (required compatibility with Java 11)
          - Variant 'releaseRuntimePublication' capability no.nordicsemi.android.support.v18:scanner:1.5.0 declares a runtime of a library, and its dependencies declared externally:
              - Incompatible because this component declares a component, with the library elements 'aar' and the consumer needed a component, packaged as a jar
              - Other compatible attribute:
                  - Doesn't say anything about its target Java version (required compatibility with Java 11)

```
Setting the version back to 1.4.3 fixes the problem. Do you have any ideas how to get it working with version 1.5.0?

philips77, 2021-06-23:
Include 
```groovy
repositories {
   mavenCentral()
}
```
in your main **build.gradle** file. 1.4.3 was the last version published to JCenter before it went down.

philips77, 2021-06-23:
Example: https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/commit/11c9c8469d3a6399541e4a50e7998b94cb6f1851

philips77, 2021-06-23:
Also, remove `jcenter()` repository.

closed 2021-06-23 by philips77

NanaDada, 2021-06-23:
Sorry, I should have mentioned it earlier: jcenter() has already been replaced by mavenCentral(). Do you have any other ideas what could be the cause, or how to fix/workaround it?

philips77, 2021-06-23:
Than i think the solution lies in:

> Incompatible because this component declares a component, with the library elements 'aar' and the consumer needed a component, packaged as a jar

How do you include the dependency?

philips77, 2021-06-23:
Something like https://stackoverflow.com/a/63486990 ?