nordicsemi/Android-Scanner-Compat-Library #106: Scanner broken when targeting Android 12+

jonschmidt, 2021-08-06:
This looks to be preventing us from targeting 12 at the moment.
```
 java.lang.IllegalArgumentException: co.dance.rider: Targeting S+ (version 31 and above) requires that one of FLAG_IMMUTABLE or FLAG_MUTABLE be specified when creating a PendingIntent.
    Strongly consider using FLAG_IMMUTABLE, only use FLAG_MUTABLE if some functionality depends on the PendingIntent being mutable, e.g. if it needs to be used with inline replies or bubbles.
        at android.app.PendingIntent.checkFlags(PendingIntent.java:375)
        at android.app.PendingIntent.getBroadcastAsUser(PendingIntent.java:645)
        at android.app.PendingIntent.getBroadcast(PendingIntent.java:632)
        at no.nordicsemi.android.support.v18.scanner.BluetoothLeScannerImplOreo.createStartingPendingIntent(BluetoothLeScannerImplOreo.java:171)
```

philips77, 2021-08-06:
Could you create a PR, please?

roum, 2021-08-25:
PR https://github.com/NordicSemiconductor/Android-Scanner-Compat-Library/pull/108

jonschmidt, 2021-09-23:
Hi. I apologize for not responding to the PR request, but I did not see the notification. Thanks @roum for the assist.

Can I ask when this will be released @philips77 ? This is a pretty major problem for us, and I fear I will have to deploy this privately to get the fix anytime soon.

philips77, 2021-09-23:
Hi,
Sorry for the delay. I'll merge it as soon as possible.

philips77, 2021-09-29:
Merged, I'll do a release today.

roum, 2021-09-30:
@philips77 I've been testing the changes in #108 on our fork, and though it does prevent fatal crashes on Android 12+ caused by the mutability flag not being explicitly set, it now prevents scan results from being received in `PendingIntentReceiver` (i.e. the intent received has no extras). This tells me that the correct flag there may be `FLAG_MUTABLE`, which can only be used if the project targets API 31 which it currently doesn't.

This issue may need to be reopened, and hopefully inspire a deeper look into how the project can safely target 31.

philips77, 2021-10-01:
I'll take a look at it. Thank you for testing!

philips77, 2021-10-01:
Right. The system needs to update the `PendingIntent` with discovered devices.

I'll set the *targetSdkVersion* to 31 for the next version and will add `FLAG_MUTABLE`.

closed 2021-10-01 by philips77

philips77, 2021-10-06:
Version 1.6.0 is just being released. Hope it will work for you!

PR #108 (2021-09-29): Fix #106
This PR fixes #106. The IMMUTABLE flags are added to `PendingIntent`s.
commit cc228266 (2021-09-29): Merge pull request #108 from barkinglabs/bugfix/use-flag-immutable

Fix #106
commit 41e9afe1 (2021-10-01): Revert "Fix #106"